energia/Energia #544: c2000 on Mac OSX

jmgc, 2015-01-15:
I have tried to upload the an program with the master branch of Energia to solve the issue #531 that I have build and the latest drivers for Mac OS X issue #500. Now I have found a new problem with the serial loader. First, the serial_loader2000 mod had to be changed to executable as the ant run did not change it.

After that I have tried to upload the device. By default, Energia uses the /dev/tty.usbserial-TI, and it simply gets stucked. So I have tried to use from the command line, using the /dev/cu.usbserial-TI, as shown below. In this case, the message Downloading appears, but it does not finish.

serial_loader2000 -f /var/folders/f2/hhqw1wkn2lv1cbprbnvvdrs00000gn/T/build2265974080692360715.tmp/Blink.cpp.txt -k Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/F28027_flash_kernel/Debug/flash_kernel.txt -b 9600 -p /dev/cu.usbserial-TIY0BLS6B

C2000 Serial Firmware Upgrader
Copyright (c) 2013 Texas Instruments Incorporated.  All rights reserved.

Downloading /var/folders/f2/hhqw1wkn2lv1cbprbnvvdrs00000gn/T/build2265974080692360715.tmp/Blink.cpp.txt to device...
Kernel auto baud timed out

In all cases I have modified the state of the switch S1 to on-off-off, the S4 is also on and I have maintained the reset button pressed. Does any one have a hint on how to solve it?

Thanks in advance

jmgc, 2015-01-19:
I have checked the schematics and the correct serial port is the /dev/tty.usbserial-TI*B. Also I have followed the procedure to press the reset button, press the ok button on the screen and once the blue led of TX is illuminated, I released the reset. Then it has worked perfectly well.

closed 2015-01-19 by jmgc

rei-vilo, 2015-01-23:
The procedure runs fine up to 38400 bauds. Faster speeds end with an error.

>Unable to open establish kernel autobaud