energia/Energia #545: CC3200 Server client.write sometimes sends incorrect data

spirilis, 2015-01-16:
A slight twist on the traditional SimpleWiFiWebServer revealed some odd behavior where client.print() output sometimes gets mangled if the length of a single string passed to client.print() exceeds 100 bytes.  Note this can be somewhat adjusted further by what has been written beforehand.  Modifying WiFiClient::write(const uint8_t *, size_t) to break up the write into <= 100 byte calls to sl_Send() seems to resolve the issue; using 101-byte calls exhibits the problem.

A test sketch to elucidate this problem:
```
#include <WiFi.h>
#include <WiFiClient.h>
#include <SPI.h>

const char ssid[] = "energia";
const char wifipw[] = "password";

WiFiServer server(80);

void setup()
{
  uint32_t msave;
  int i;
  
  Serial.begin(115200);
  
  delay(2000);
  Serial.println("WiFi:");
  WiFi.begin((char *)ssid, (char *)wifipw);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(150);
  }
  Serial.println();
  
  Serial.print("Connected- waiting for IP..");
  msave = millis();
  while (WiFi.localIP() == INADDR_NONE) {
    Serial.print('.');
    delay(150);
    if (millis()-msave > 15000) {
      Serial.println("Timeout waiting for IP; disconnecting & reconnecting.");
      WiFi.disconnect();
      WiFi.begin((char *)ssid, (char *)wifipw);
      msave = millis();
    }
  }
  Serial.print(" IP=");
  Serial.println(WiFi.localIP());
  
  //Serial.println("Setting power management to Long-Sleep Interval 1200ms-");
  //WiFi.setPowerMgmt(WL_POWER_LONGSLEEP, 1200);
  
  Serial.println("Starting Webserver on port 80-");
  server.begin();
}

void loop()
{
  int i = 0;
  static unsigned int ttlclients = 0;
  WiFiClient client = server.available();
  
  if (client) {
    Serial.print("Client #"); Serial.println(ttlclients++);
    
    char buffer[2048];
    memset(buffer, 0, 2048);
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        if (c == '\n') {
          if (strlen(buffer) == 0) {
            client.print("HTTP/1.1 200 OK\r\n");
            client.print("Content-type: text/html\r\n\r\n");
            //client.println();
            // wtf, sometimes client.print's don't send everything?
            client.println("<html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>");
            client.print('\r'); client.print('\n');
            //delay(50);
            client.print("<body align=center><h1 align=center><font color=\"red\">Welcome to the CC3200 WiFi Web Server</font></h1>\r\n");
            client.print("RED LED <button onclick=\"location.href='/H'\">HIGH</button>");
            client.print("<button onclick=\"location.href='/L'\">LOW</button><br>\r\n");
            
            client.print("</body></html>\r\n\r\n");
            //client.println();
            // break out of while loop:
            break;
          } else {
            memset(buffer, 0, 2048);
            i = 0;
          }
        } else if (c != '\r') {
          buffer[i++] = c;
        }
        
        // Check to see if client request was GET /H or GET /L
        if (endsWith(buffer, "GET /H")) {
          digitalWrite(RED_LED, HIGH);
        }
        if (endsWith(buffer, "GET /L")) {
          digitalWrite(RED_LED, LOW);
        }
      } /* client.available() */
    }
    // Close connection
    client.stop();
    Serial.println("Client disconnected from local host.");
  }
}

//
//a way to check if one array ends with another array
//
boolean endsWith(char* inString, char* compString) {
  int compLength = strlen(compString);
  int strLength = strlen(inString);
  
  //compare the last "compLength" values of the inString
  int i;
  for (i = 0; i < compLength; i++) {
    char a = inString[(strLength - 1) - i];
    char b = compString[(compLength - 1) - i];
    if (a != b) {
      return false;
    }
  }
  return true;
}
```

Using the CLI with telnet under Linux for port 80 access, the problem looks like this:
```
[ebrundic@spock ~]$ telnet 192.168.0.195 80
Trying 192.168.0.195...
Connected to 192.168.0.195.
Escape character is '^]'.
GET / HTTP/1.0

HTTP/1.1 200 OK
Content-type: text/html

<html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>

dy align=center><h1 align=center><font color="red">Welcome to the CC3200 WiFi We1>Server</font></h1>
RED LED <button onclick="location.href='/H'">HIGH</button><button onclick="location.href='/L'">LOW</button><br>
</body></html>

Connection closed by foreign host.
[ebrundic@spock ~]$ 
```
Note the missing "<bo" characters; instead the line begins with "dy align=center".  This should read "<body align=center".
I have debugging hooks enabled with WiFiClient::write() that show me what is going on at its level; everything looks peachy, the sl_Send() call is being executed with the correct data and the value of iRet returned matches the size of the string sent.  The error presumably happens either within the SimpleLinkWiFi API or within the NWP firmware itself.

Performing a hook that limits WiFiClient::write() to 100-byte increments resolves the issue.  More on that in a bit after further testing.

spirilis, 2015-01-16:
Example WiFiClient::write(const uint8_t *, size_t) which allows tweaking of the max write size and adds Serial debugging messages:
```
size_t WiFiClient::write(const uint8_t *buffer, size_t size)
{
    //
    //don't do anything if not properly set up
    //
    if (_socketIndex == NO_SOCKET_AVAIL) {
        return 0;
    }

    size_t origsize = size;
    boolean did_recurse = false;

    // Limit to 100-byte writes
    #define WIFICLIENT_WRITE_INCREMENT 101
    uint8_t *bptr = (uint8_t *)buffer;
    while (size > WIFICLIENT_WRITE_INCREMENT) {
        did_recurse = true;
        if (write((const uint8_t *)bptr, WIFICLIENT_WRITE_INCREMENT) == 0)
            return 0;  // Error/failure, client already stopped, just bail
        bptr += WIFICLIENT_WRITE_INCREMENT;
        size -= WIFICLIENT_WRITE_INCREMENT;
    }
    //
    //write the buffer to the socket
    //
    Serial.print(">> sl_Send("); Serial.print(size); Serial.print(")"); Serial.print(' '); Serial.write((const uint8_t *)bptr, size);
    int iRet = sl_Send(WiFiClass::_handleArray[_socketIndex], (const uint8_t *)bptr, size, NULL);

    // Flow control signal; perform a paced-retry.
    while (iRet == SL_EAGAIN) {
        Serial.print(" #");
        delay(10);
        iRet = sl_Send(WiFiClass::_handleArray[_socketIndex], (const uint8_t *)bptr, size, NULL);
    }

    if ((iRet < 0) || (iRet != size)) {
        //
        //if an error occured or the socket has died, call stop()
        //to make the object aware that it's dead
        //
        Serial.println(" ERR");
        stop();
        return 0;
    } else {
        Serial.print("; iRet="); Serial.println(iRet);
        if (did_recurse)
            return origsize;
        return iRet;
    }
}
```

With that code (and #define WIFICLIENT_WRITE_INCREMENT set to 101, instead of 100) we get this from the serial console:
```
WiFi:
...
Connected- waiting for IP............................... IP=192.168.0.195
Starting Webserver on port 80-
Client #0
GET / HTTP/1.0

>> sl_Send(17) HTTP/1.1 200 OK
; iRet=17
>> sl_Send(27) Content-type: text/html

; iRet=27
>> sl_Send(85) <html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>; iRet=85
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(101) <body align=center><h1 align=center><font color="red">Welcome to the CC3200 WiFi Web Server</font></h; iRet=101
>> sl_Send(4) 1>
; iRet=4
>> sl_Send(58) RED LED <button onclick="location.href='/H'">HIGH</button>; iRet=58
>> sl_Send(55) <button onclick="location.href='/L'">LOW</button><br>
; iRet=55
>> sl_Send(18) </body></html>

; iRet=18
Client disconnected from local host.
```

Clearly the whole "<body ... blahblah" string was written correctly, but the output from telnet says not:
```
[ebrundic@spock ~]$ telnet 192.168.0.195 80
Trying 192.168.0.195...
Connected to 192.168.0.195.
Escape character is '^]'.
GET / HTTP/1.0

HTTP/1.1 200 OK
Content-type: text/html

<html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>

dy align=center><h1 align=center><font color="red">Welcome to the CC3200 WiFi We1>Server</font></h1>
RED LED <button onclick="location.href='/H'">HIGH</button><button onclick="location.href='/L'">LOW</button><br>
</body></html>

Connection closed by foreign host.
```
I have analyzed this with Wireshark once before and the bug shows there in the packet flow as expected.  Smells like a bug in the NWP or API as mentioned.

spirilis, 2015-01-16:
PS- Change the #define WIFICLIENT_WRITE_INCREMENT to 100, and we get this on the serial console:
```
WiFi:
...
Connected- waiting for IP....................................... IP=192.168.0.195
Starting Webserver on port 80-
Client #0
GET / HTTP/1.0

>> sl_Send(17) HTTP/1.1 200 OK
; iRet=17
>> sl_Send(27) Content-type: text/html

; iRet=27
>> sl_Send(85) <html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>; iRet=85
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(1) 
; iRet=1
>> sl_Send(100) <body align=center><h1 align=center><font color="red">Welcome to the CC3200 WiFi Web Server</font></; iRet=100
>> sl_Send(5) h1>
; iRet=5
>> sl_Send(58) RED LED <button onclick="location.href='/H'">HIGH</button>; iRet=58
>> sl_Send(55) <button onclick="location.href='/L'">LOW</button><br>
; iRet=55
>> sl_Send(18) </body></html>

; iRet=18
Client disconnected from local host.
```

And this from the CLI telnet session:
```
[ebrundic@spock ~]$ telnet 192.168.0.195 80
Trying 192.168.0.195...
Connected to 192.168.0.195.
Escape character is '^]'.
GET / HTTP/1.0

HTTP/1.1 200 OK
Content-type: text/html

<html><head><title>Energia CC3200 WiFi Webserver with Power Management</title></head>

<body align=center><h1 align=center><font color="red">Welcome to the CC3200 WiFi Web Server</font></h1>
RED LED <button onclick="location.href='/H'">HIGH</button><button onclick="location.href='/L'">LOW</button><br>
</body></html>

Connection closed by foreign host.
[ebrundic@spock ~]$ 
```

Looks correct there, the body line starts with "<body align=center" as we would expect.

spirilis, 2015-01-16:
Fwiw- I tried changing that #define to 64, 96, 100, 101, 102, 104 and it seems that 100 is the magic "maximum", beyond which the bug shows up.

robertinant, 2015-01-16:
Thanks @spirilis for the detailed analysis and catching this bug! I will get with the Simpelink CC3200 team to figure out what the issue is. To make sure, what is the version of the service pack on your CC3200?

spirilis, 2015-01-16:
@robertinant Does this have what you need version-wise?
```
Open UART /dev/ttyUSB1
open UART success
Getting storage list
Bootloader Version: 3
Silicon version ES1.32 or higher
Bootloader version is 2, 0, 3, 2
It's a CC3200 device: PG1.32
BlockSize is 4096, number of blocks is 64
erasing 1 blocks starting from  4
Switch to NWP bootloader complete
Load common boot command for PG1.32
Bootloader version is 2, 0, 3, 4
```
I believe I haven't updated it ever since the initial update back in August.

spirilis, 2015-01-16:
PS- Problem confirmed in AP mode too.

spirilis, 2015-01-16:
Can't confirm the bug on CC3100 w/ TM4C123 FYI.  Even with an unpatched WiFiClient.cpp, it works fine.

spirilis, 2015-01-18:
@robertinant Would you like to see a pull request with my workaround or just wait on the CC3200 team's results?

pagnamen, 2015-02-03:
@spirilis 

I have the same problem.  The source of the web page in firefox does not correspond to what I send with client.print. There often are missing characters on the beginning of some lines.
Putting some delays everywhere partially solved the problem but still I have random format issues.
Moreover, when I hit refresh on the browser or when I manipulate the sliders on the webpage, the CC3200 randomly crashes.
Implementing your solution did not change anything (unfortunately).
I will try to import the project in CCS and have a look with the debugger.
Let me know if there is some news/advise on this issue.

robertinant, 2015-02-03:
@spirilis, can you send me a pull request with the work-around? It seems that we might have to work around it for now.