energia/Energia #551: TRF7970A NFC Boosterpack Energia Library

avinashchakravarthi, 2015-01-18:
hi,frnds
  i would like to work with msp430g2553 and nfcboosterpack trf7970a in energia IDE .Does any one have energia library for TRF7970A boosterpack. 
 
i have attached a pdf which helps to make a library for iso1443a and iso5693
(in this link ---->http://forum.43oh.com/topic/6304-trf7970abp-nfc-boosterpack-enegia-library/)
 
i tried this code for msp430g2553 but its not working please any one can help me

#include<SPI.h>

#define enable P2_2
#define CS P2_1
#define interrupt P2_0
#define CS_ENABLED digitalWrite(CS,LOW)
#define CS_DISABLED digitalWrite(CS,HIGH)

void setup()
{
  // put your setup code here, to run once:
  pinMode(CS,OUTPUT);
  pinMode(enable,OUTPUT);
  digitalWrite(enable,LOW);
  pinMode(interrupt,INPUT);
  SPI.setDataMode(SPI_MODE1);
  SPI.setClockDivider(SPI_CLOCK_DIV8);
  SPI.setBitOrder(MSBFIRST);
  SPI.begin();
  Serial.begin(9600);
}

void loop()
{
  // put your main code here, to run repeatedly:
 

   let();
}
void let(void)
{
  CS_ENABLED;
  Write(0x83);
  CS_DISABLED;
  CS_ENABLED;
  Write(0x09);
   Write(0x31);
  CS_DISABLED;
  CS_ENABLED;
  Read(0x49);
  CS_DISABLED;
  CS_ENABLED;
   Write(0x01);
   Write(0x88);
  CS_DISABLED;
   CS_ENABLED;
  Read(0x41);
  CS_DISABLED;
   CS_ENABLED;
   Write(0x00);
   Write(0x20);
  CS_DISABLED;
  CS_ENABLED; 
  Read(0x40);
  CS_DISABLED;
  CS_ENABLED;
 Write(0x01);
 Write(0x88);
CS_DISABLED;
CS_ENABLED;
 Write(0x8F);
 Write(0x90);
 Write(0x3D);
 Write(0x00);
 Write(0x0F);
 Write(0x26);
CS_DISABLED;
delay(100);
CS_ENABLED;
 Write(0x6C);
Read(0xC0);
Read(0x3E);
CS_DISABLED;
CS_ENABLED;
 Write(0x5C);
Read(0x02);
CS_DISABLED;
CS_ENABLED;
 Write(0x7F);
Read(0x04); 
Read(0x00);
CS_DISABLED;
CS_ENABLED;
 Write(0x8F);
CS_DISABLED;
CS_ENABLED;
SPI.transfer(0x4F);
Read(0x7F);
CS_DISABLED;
delay(20);
CS_ENABLED;
 Write(0x8F);
 Write(0x90);
 Write(0x3D);
 Write(0x00);
 Write(0x20);
 Write(0x93);
 Write(0x20);
CS_DISABLED;
delay(5);
CS_ENABLED;
 Write(0x6C);
Read(0xC0);
Read(0x3E);
CS_DISABLED;
CS_ENABLED;
 Write(0x5C);
Read(0x05);
CS_DISABLED;
CS_ENABLED;
 Write(0x7F);
Read(0xCD); 
Read(0x88); 
Read(0x77); 
Read(0x66); 
Read(0x00);
delay(10);
CS_DISABLED;
CS_ENABLED;
 Write(0x6C);
Read(0x00); 
Read(0x3E);
CS_DISABLED;
CS_ENABLED;
 Write(0x8F);
CS_DISABLED;
CS_ENABLED;
 Write(0x4F);
Read(0x7F);
CS_DISABLED;
}

void Read(byte data )
{
  Serial.print(SPI.transfer(data));
}

void Write(byte data)
{
 SPI.transfer( data);
}

spirilis, 2015-01-18:
I don't know if anyone has done much with NFC and Energia yet.  I own the TRF boosterpack and the RF430CL330H ones but haven't yet had the time to tinker.  TRF's datasheet looked formidable at first glance a while back... but it is supposed to be a jack-of-all-trades chipset.

spirilis, 2015-02-18:
Closing for now ... there's already discussions on 43oh about this

closed 2015-02-18 by spirilis