energia/Energia #553: CC3200 from AP to STA

PaoloIT, 2015-01-18:
Hello everybody,
I need help to find a way to pass from AP mode to STA mode in CC3200.
My system start in AP mode with a webserver. Here I can set SSID and password to connect the board to my router in STA mode. After the setting I tried to connect to router in STA mode but system return WL_CONNECT_FAILED at WiFi.begin command. 
If i try to use WiFi.begin from the main of firmware and everything work fine, after WiFi.beginNetwork no.
I tried to use WiFi.disconect, sl_Stop etc... but nothing... Who can help me?

rei-vilo, 2015-01-18:
You'll receive more answers if you post your question at http://forum.43oh.com.

closed 2015-01-18 by rei-vilo

Agam-aviconn, 2017-01-31:
#ifndef __CC3200R1M1RGC__
#include <SPI.h>
#endif
#include <WiFi.h>
#include <udma_if.h>
const char ssid[] = "CC3200";
const char wifipw[] = "password";

const char s[2] = "-";
char *token;

String y = "";
char Rssid[100],Rpsk[100];

unsigned int Port =2390;
char packetBuffer[255];
IPAddress Ip(255,255,255,255);
char d_name[]="Device Name:Texas Instruments CC3200 Board,\n";
char d_id[]="Device Id:1,\n";
char d_type[]="Device type:Controller with Wi-Fi,\n";
char  ReplyBuffer[] = "Device IP is:";
char b[1024];
int j=0;

WiFiUDP Udp;

void setup()
{
  Serial.begin(115200);
  pinMode(RED_LED, OUTPUT);  // LED will toggle when clients connect/disconnect
  digitalWrite(RED_LED, LOW);
  
  Serial.print("Setting up Access Point named: ");
  Serial.println(ssid);
  Serial.print("AP uses WPA and password is: ");
  Serial.println(wifipw);
  
  WiFi.beginNetwork((char *)ssid, (char *)wifipw);

  while (WiFi.localIP() == INADDR_NONE) 
  {
    // print dots while we wait for the AP config to complete
    Serial.print('.');
    delay(300);
  }
  Serial.println();
  Serial.println("AP active.");
  printWifiStatus();
  Udp.begin(Port);
}

unsigned int num_clients = 0;

void loop()
{
  unsigned int a, i;

  a = WiFi.getTotalDevices();
  

  // Did a client connect/disconnect since the last time we checked?
  if (a != num_clients) 
  {
    if (a > num_clients) 
    {  // Client connect
      digitalWrite(RED_LED, !digitalRead(RED_LED));
      Serial.println("Client connected! All clients:");
      for (i = 0; i < a; i++) 
      {
        Serial.print("Client #");
        Serial.print(i);
        Serial.print(" at IP address = ");
        Serial.print(WiFi.deviceIpAddress(i));
        Serial.print(", MAC = ");
        Serial.println(WiFi.deviceMacAddress(i));
      }
    } 
    else 
    {  // Client disconnect
      digitalWrite(RED_LED, !digitalRead(RED_LED));
      Serial.println("Client disconnected.");
    }    
    num_clients = a;
  }
  
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    int len = Udp.read(packetBuffer, 255);
    if (len > 0)packetBuffer[len] = 0;
    //unit=atoi(packetBuffer);
    Serial.println(packetBuffer);
         
                 /* get the first token */
                token = strtok(packetBuffer, s);
                int w = 0;
   
                while( token != NULL ) 
               {
                  
                  if(w == 0){
                  strcpy(Rssid,token);
                  Serial.println(Rssid);
                  w++;
                  }else if(w == 1){
                    strcpy(Rpsk,token);
                    Serial.println(Rpsk);
                  }
                  token = strtok(NULL, s);
               }

               /* int retVal;
                retVal = sl_WlanSetMode(ROLE_STA);
                retVal = sl_Stop(30);
                retVal= sl_Start(NULL, NULL, NULL);
               */
               UDMAInit();
  sl_Start(NULL, NULL, NULL);

  sl_WlanDisconnect();
  
  sl_NetAppMDNSUnRegisterService(0, 0);
  
  sl_WlanRxStatStart();
  
  sl_WlanSetMode(ROLE_STA);

  /* Restart Network processor */
  sl_Stop(30);

  sl_Start(NULL, NULL, NULL);

                // attempt to connect to Wifi network:
                Serial.print("Attempting to connect to Network named: ");
                delay(5000);
                // print the network name (SSID);
                Serial.println(Rssid);
                // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
                WiFi.begin(Rssid, Rpsk);
                while ( WiFi.status() != WL_CONNECTED) {
                  // print dots while we wait to connect
                  Serial.print(".");
                  delay(300);
                }

                Serial.println("\nYou're connected to the network");
                Serial.println("Waiting for an ip address");

                while (WiFi.localIP() == INADDR_NONE) {
                  // print dots while we wait for an ip addresss
                  Serial.print(".");
                  delay(300);
                }

                Serial.println("\nIP Address obtained");

                // you're connected now, so print out the status
                printWifiStatus();
   }

}

void printWifiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());
  IPAddress ip = WiFi.localIP();
  sprintf(b,"%d.%d.%d.%d,",ip[0],ip[1],ip[2],ip[3] );
  Serial.print("IP Address: ");
  Serial.println(ip);
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}