energia/Energia #554: LAUNCHXP-F28027 — No Serial Port on Mac OS X 10.10 — Driver Needed?

rei-vilo, 2015-01-19:
Even with the correct configuration —**S1** `on-on-on` and **S4** `on`— no serial connection appears on Mac OS X 10.10 *Yosemite*.

Pressing **S3** to launch the demo as described in LaunchPad F28027F Started (sprz413), the TX LED blinks but nothing is received on the Mac.

Both `ls /dev/cu.*` and `ls /dev/tty.*` return nothing.

Is a driver needed for Mac OS X 10.10 *Yosemite*?

robertinant, 2015-01-19:
A driver is needed for this LaunchPad: Signed driver is in this issue https://github.com/energia/Energia/issues/500. Will post this driver on energia.nu later this week.

rei-vilo, 2015-01-19:
Thanks! So the same driver handles the CC3200 and the C2000.

closed 2015-01-19 by rei-vilo