energia/Energia #555: Button example code/comments wrong

ecxr, 2015-01-25:
Examples->Digital->Button indicates the LED will light up when the button is pressed when the buttonState is HIGH.  I guess the pullup characteristics of the launchpad are different from the arduino example this was ported from.  

In fact, the LED will be OFF when buttonState is HIGH and ON when buttonState is LOW.

Either the comments should be fixed to reflect the current implementation, or the code/comments fixed to indicate the LED will be ON when buttonState is LOW

  // check if the pushbutton is pressed.
  // if it is, the buttonState is HIGH:
  if (buttonState == LOW) {     
    // turn LED on:

rei-vilo, 2015-01-25:
Which board please?

ecxr, 2015-01-25:
MSP430G2553/Rev 1.5

rwerthman, 2017-09-28:
This still hasn't been fixed.  The same problem occurs with my MSP430F5529LP when I use Energia 1.6.10E18.  It appears based on tutorial http://embedded.fm/blog/ese101-schematics-pullups I've been following that when the button is not pushed and has a pull-up resistor the voltage is high.  When the button is pushed the the switch is closed and the button is connected to ground with 0 voltage.  So the button is high when not pushed and low when pushed.

I wouldn't mind doing my first open source bug fix and pull request fixing this issue.

rwerthman, 2022-02-17:
Hello @ecxr and @rei-vilo,

I abandoned my pull request (https://github.com/energia/Energia/pull/987) to fix this since I no longer have the hardware to test on.