energia/Energia #557: "DEBUG cc3200prog.c:<line_number>:write failed" after download complete

XuGuohui, 2015-01-28:
Hi,
When I upload a sketch to CC3200 using Energia, after download complete, the console still print the message as below:

Downloading file "/sys/mcuimg.bin" with size 4064
.Download complete
DEBUG cc3200prog.c:2623: write failed
DEBUG cc3200prog.c:2629: write failed
DEBUG cc3200prog.c:2635: write failed
DEBUG cc3200prog.c:2641: write failed

Does anyone know why it causes the additional messages printed on the console and what it means? Thanks in advance!

rei-vilo, 2015-01-31:
Are the pins configured properly? Please check http://energia.nu/pin-maps/guide_cc3200launchpad/

What is the exact reference of your CC3200 LaunchPad?

robertinant, 2015-02-02:
What version of LaunchPad do you have? Does the firmware run and work as expected regardless of teh write fail? Can you post the entire output of the console please? This will help a lot with determining what is going wrong.

I just looked at what the lines of code do at it has to do with writing to the FTDI chip in GPIO mode. Have you installed the FTDI drivers from http://energia.nu/guide/guide_windows/ ?

XuGuohui, 2015-02-06:
Hi @rei-vilo , @robertinant ,
I was using the released CC3200 module R1M2 on our own board. And the USB-UART bridge we are using is a ARM MCU running the CMSIS DAP firmware http://developer.mbed.org/handbook/cmsis-dap-interface-firmware , instead of the FTDI. We can automatically reset the CC3200 mudule and force it to run the UART bootloader when it detect a break signal from Energia IDE. So the sketch can upload to CC3200 but with some additional messages:

Opening \\.\COM98
Getting storage list
Bootloader Version: 4
Silicon version ES1.32 or higher
Bootloader version is 2, 1, 4, 0
It's a CC3200 device: PG1.33 or higher
Switch UART pinmux to APPS
Switch to NWP bootloader complete
Load common boot command for PG1.33 or higher
Bootloader version is 2, 0, 4, 0
BlockSize is 4096, number of blocks is 16
erasing 13 blocks starting from  0
erasing file "/sys/mcuimg.bin"
deleting file "/sys/mcuimg.bin"
erase file completed
Downloading file "/sys/mcuimg.bin" with size 4064
.Download complete
DEBUG cc3200prog.c:2623: write failed
DEBUG cc3200prog.c:2629: write failed
DEBUG cc3200prog.c:2635: write failed
DEBUG cc3200prog.c:2641: write failed
Can't open device

It seams that something activates the debug procedure. Any idea would appreciate!

flety-emmanuel, 2015-03-25:
Greetings. I'm having the same exact issues, using a CC3200 MOD(ule) version. I'm probably going to open another discussion as I have other concerns about 0013 vs 0014 issues, various code problems and no possibility to properly execute code. 
Any help regarding the loader would be appreciated. There's a growing number of the MOD users vs the launch pad, and I think there are a few specifics to solve for the module.

robertinant, 2015-03-25:
The reason for this is that the cc3200prog expects an FTDI chip. It uses the second port of the FTDI chip to toggle the SOP2 after upload has been done so that the firmware starts without having to reset the board and remove the SOP2 jumper. There is very little I can do to support custom boards in cc3200prog since all of them will be different. Not sure if either of you works for RedBearLab but I am working with them to add support for their version of the module and Arduino style board.

flety-emmanuel, 2015-03-25:
I totally got that and I'm currently looking at options to have a more automatic upload on the board, however the problem is different here: I (we ?) automatically trigger the bootloader with a manual reset and control of the SOP2 pin so that it's going in bootloader mode. Upload looks like it's working. My board is also working fine using uniflash, so the dialog with the chip bootloader is working ok but I'm unsure about what the debug info means. I'm going to see with readbear labs to see what can be done about it. Indeed, a simple control thru the DSR or DTS pins and a coupling capacitor could allow a very nice automatic reset + flash system. I don't think most of use will ever implement the FT2232 in our design, I personally don't care about jtag for my platform. Thanks for the answer !

davoine, 2015-04-25:
Indeed, RedBearLab Wifi Mini does have this problem when using Energia. It would be great to solve it :)

flety-emmanuel, 2015-04-27:
thanks for confirming this. On my design, my auto reset works just fine, I just have to manually control SOP2 but aside of that, I can program my board normally. Having the source code or custom versions of CC3200 prog would be great. While I think the code comes from RBL, it's been probably customized by Energia people to match the launch pad. Come on people, release that so we can compile our own, to be free of choosing which pin we use to jump to flash mode !!! thanks

BJSingh, 2015-05-02:
I am too facing same upload issue with RedBearLab Wifi Mini when using with Energia. I have also attach upload error in Win7 and Win8.1 OS.
[image]

[image]

omh11, 2016-11-29:
I am still having this issue though with the RedBearLab CC3200. Has there been any updates @robertinant ?

aarongamble, 2017-02-03:
I found that Energeria 17 works to upload sketches and reset the board. Energeria 18 will upload sketches, but you need to do a manual reset after seeing the 'write failed' messages.