energia/Energia #558: java.lang.IllegalArgumentException: scrollCount and interval must be greater than 0

spirilis, 2015-01-29:
So for some odd reason, my Energia 14 install just magically started crashing on startup throwing this error.  It seems anytime I have local libraries installed this error happens, and I'm not sure why e.g. it wasn't happening this morning but it's happening 100% of the time now, even with a fresh reinstall of Energia.

Full trace:
```
Exception in thread "main" java.lang.IllegalArgumentException: scrollCount and interval must be greater than 0
	at processing.app.tools.MenuScroller.<init>(MenuScroller.java:310)
	at processing.app.tools.MenuScroller.<init>(MenuScroller.java:283)
	at processing.app.tools.MenuScroller.setScrollerFor(MenuScroller.java:166)
	at processing.app.Editor.rebuildExamplesMenu(Editor.java:632)
	at processing.app.Editor.buildFileMenu(Editor.java:513)
	at processing.app.Editor.buildMenuBar(Editor.java:475)
	at processing.app.Editor.<init>(Editor.java:206)
	at processing.app.Base.handleOpen(Base.java:768)
	at processing.app.Base.handleOpen(Base.java:733)
	at processing.app.Base.handleNew(Base.java:584)
	at processing.app.Base.<init>(Base.java:321)
	at processing.app.Base.main(Base.java:201)
```

However, addressing the basic error, I am puzzled why app/src/processing/app/Editor.java line 632 has:
```
      MenuScroller.setScrollerFor(examplesMenu,-1,-1,upper>0?upper+1:0,lower);
```

When that function clearly disallows scrollCount and intervals <= 0, per line ~310ish of tools/MenuScroller.java:
```
  public MenuScroller(JPopupMenu menu, int scrollCount, int interval,
          int topFixedCount, int bottomFixedCount) {

    if(scrollCount == -1)
        scrollCount = getMaximumItems()-topFixedCount-bottomFixedCount; // Autosize

    if(interval == -1)
        interval = 150; // Default value

    if (scrollCount <= 0 || interval <= 0) {
      throw new IllegalArgumentException("scrollCount and interval must be greater than 0");
    }
```

Is there any good reason why Energia is running rebuildExamplesMenu now vs. not doing it this morning (I'm assuming that's the case here), and why is it calling the MenuScroller with such invalid parameters?

What's even more bizarre is I haven't added any libraries recently!

spirilis, 2015-01-30:
FYI, my Energia 13 install starts correctly.

spirilis, 2015-01-30:
Ok, after editing ~/.energia/preferences.txt and changing:

board from lpmsp430g2231 to lptm4c1230c3pm
and target from msp430 to lm4f

I can start Energia again with stuff inside my libraries/ local folder.
I might not have actually tried using MSP430 stuff with Energia 14 until just yesterday.

spirilis, 2015-01-30:
Sure enough, I can start Energia with preferences.txt pointing to the Tiva, then switch to MSP430F5529 and do a compile.

But after exiting Energia and restarting, I get those Java exceptions again and no GUI (terminal used to start energia just hangs until I CTRL-C it.)

robertinant, 2015-02-02:
I have seen this as well. I will look into finding a way to reliably reproduce it so that I can debug it.
rebuildExamplesMenu should run everytime you change board. This is so that the examples and library menu's get populated with board specific library examples.

spirilis, 2015-02-03:
Goofy test-

cd energia-0101E0014/hardware/msp430/libraries
(back up the libraries folder for kicks)

find . -name examples -exec rm -Rf {} \;

Deletes all examples from all "stock" libraries that ship with energia...
And then it runs fine.

spirilis, 2015-02-03:
Restoring the libraries folder, then deleting the examples/ subfolder one by one...

```
[ebrundic@spock libraries]$ for i in * ; do
> echo $i -
> rm -Rf ${i}/examples
> read BLAH
> done
Adafruit_TMP006 -

AIR430BoostEuropeETSI -
    
AIR430BoostUSAFCC -

aJson -
```

Boom, after the fourth library folder's examples subfolder is deleted, Energia starts correctly.
Suspecting maybe aJson is messing it up, I restored the libraries folder and then deleted aJson/examples by itself.  Energia wouldn't start.

So I restored it again, and this time deleted four library folders' worth of examples - aJson/examples, AIR430BoostUSAFCC/examples (still wouldn't start), AIR430BoostEuropeETSI/examples (still wouldn't start), then rather than repeat the first test, I selected the next one in the list - CogLCD/examples.  Then Energia started.

I suspect maybe there's a limit on the maximum # of library folders allowed to have examples?
I tried deleting the examples/ folder for one of my local ~/Documents/Energia/libraries library folders, then re-tested my stock msp430/libraries deletion example but it didn't seem to change.  The problem disappeared after 4 libraries' examples subfolder was deleted.

spirilis, 2015-02-03:
Also instead of deleting examples subfolders from built-in libraries at the top of the alphabetic list, I tried from the bottom:
```
[ebrundic@spock libraries]$ ls
Adafruit_TMP006        AIR430BoostUSAFCC  CogLCD     EduBPMKII_Screen  IRremote                  LiquidCrystal  MspFlash  PubSubClient  SoftwareSerial  Stepper  USBSerial  Wire
AIR430BoostEuropeETSI  aJson              DriverLib  Firmata           LCD_SharpBoosterPack_SPI  MQTTClient     PubNub    Servo         SPI             Temboo   WiFi
[ebrundic@spock libraries]$ rm -Rf Wire/examples WiFi/examples USBSerial/examples
[ebrundic@spock libraries]$ rm -Rf Temboo/examples
```

After deleting that 4th examples subfolder (Temboo/examples), Energia would start correctly.
So there's definitely something going on with the # of examples folders being allowed here.  I suspect Energia 14 added more libraries (MQTT?) so it's only starting to run up against this bug.

spirilis, 2015-02-03:
Also of note, msp430 has 23 libraries, lm4f 17, and cc3200 14.

spirilis, 2015-02-03:
In either case, the Examples menu is chock full ... and where I believe the "built-in library examples" live, even with an MSP430 selected there are only 17 libraries listed.  That arch change using base.rebuildExamplesMenu() is probably just repopulating the existing examples menu item based on its existing size, and as a result some libraries' examples go missing.

So that Examples menu needs some scrolling features too, since it doesn't scroll at all.

spirilis, 2015-02-03:
And to put things into perspective more, here's my examples menu when starting as lm4f but switching to msp430-
http://spirilis.net/junk/msp430/energia/energia-examplesmenu-size.png

4 of my "local" library examples listed in that scrolling section in the middle.  The arch-specific distribution supplied library examples down below number 17, and there are some missing from comparing against the msp430 libraries folder.

This 4-item limit is probably related to the implementation of getMaximumItems() from MenuScroller.java -
```
  /**
   * Returns the maximum items that fit in the current screen
   *
   * @author Paul Stoffregen
   * @param menu the menu
   * @return Number of items
   */
   private static int getMaximumItems() {
         JMenuItem test = new JMenuItem("test");
         ButtonUI ui = test.getUI();
         Dimension d = ui.getPreferredSize(test);
         double item_height = d.getHeight();
         JMenuItem up = new JMenuItem(MenuIcon.UP);
         ui = up.getUI();
         d = ui.getPreferredSize(up);
         double icon_height = d.getHeight();
         double menu_border_height = 8.0; // kludge - how to detect this?
         double screen_height = java.awt.Toolkit.getDefaultToolkit().getScreenSize().getHeight();
         return (int)((screen_height - icon_height * 2 - menu_border_height) / item_height);
   }
```

So on startup, Energia tries to craft the default sizes of the Examples menu, and since the arch-specific distribution supplied library examples do not get their own scrolling container, they take up so much screen height in the menu that there isn't enough space to fit my local-library example scrollbox container, so getMaximumItems() returns 0 and we get the Java exception.

BenBergman, 2015-04-02:
I'm also experiencing this issue, both in 14 and 15. I am running Arch Linux x64, however I only see this issue on one of my two machines. The other seems to work fine.

jasiek, 2015-08-24:
I'm experiencing this on both 15 and 16, running on Ubuntu 15.10 with OpenJDK 1.6

robertinant, 2015-09-25:
I will be moving Energia to the new Arduino 1.6 IDE. These issues are no longer applicable when that happens so closing the issue.

closed 2015-09-25 by robertinant