energia/Energia #559: cc3200 running multiple tcp servers possible?

holla2040, 2015-01-30:
My cc3200 project requires both telnet and http. I think there's an issue in server.available() method that makes this difficult (or I don't understand how to run multiple servers, quite possible). Here's a simple sketch showing the issue. 

#include &lt;WiFi.h&gt;

char ssid[] = "energia";
char password[] = "supersecret";

WiFiServer tcp(23);
WiFiServer http(80);

void setup() {
  Serial.begin(115200);
  Serial.print("Attempting to connect to Network named: ");
  Serial.println(ssid); 
  WiFi.begin(ssid, password);
  while ( WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println("\nYou're connected to the network");
  Serial.println("Waiting for an ip address");
  
  while (WiFi.localIP() == INADDR_NONE) {
    Serial.print(".");
    delay(300);
  }
  Serial.println("\nIP Address obtained");
  printWifiStatus();
  tcp.begin();
  http.begin();
}

void loop() {
  char c;

  WiFiClient tcpClient = tcp.available();
  if (tcpClient) {
    if (tcpClient.available() > 0) {
      c = tcpClient.read();
      tcp.write(c);
      Serial.print("tcp  ");
      Serial.println(c);
    }
  }

  WiFiClient httpClient = http.available();
  if (httpClient) {
    if (httpClient.available() > 0) {
      c = httpClient.read();
      http.write(c);
      Serial.print("http ");
      Serial.println(c);
    }
  }
}

void printWifiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
}

I then netcat to port 23 and type "this is a test", here's the serial monitor output

IP Address obtained
SSID: energia
IP Address: 192.168.0.9
http t
http h
http i
tcp  s
tcp   
http i
http s
http  
http a
tcp   
http t
http e
http s
tcp  t
tcp  

in 'loop', 

  WiFiClient tcpClient = tcp.available();
  WiFiClient httpClient = http.available();

these tcp.available() and http.available() return values aren't as expected. http.available is returning the open tcp socket. 

I looked through WiFiServer::available method and its not evaluating for port match when scanning the 8 possible connections. 

So, am I using WiFiServer correctly and available() is poorly implemented? Or is something else incorrect?

spirilis, 2015-01-30:
Sounds like it wasn't implemented with that in mind...

holla2040, 2015-01-30:
Don't think it was because in the current implementation there's no method to distinguish from the server.available(status) method return val or *status param the connection information such as port or connections src.  Nothing to dispatch on.

spirilis, 2015-01-30:
Come to think of it, I don't see much in the SimpleLinkWiFi API docs talking about what happens when you have multiple sockets in bind/listen mode.  I assume it works, it's just not spelled out in the PDF.  I haven't looked at the stock SDK examples to see if this is a covered example (most likely TI-RTOS based).

holla2040, 2015-01-31:
FYI, both server ports open and function, the available() methods just cross couples the services from both ports. I'll take a look at the api doc as well. My design kinda hinges on the assumption of multiple servers and clients functioning simultaneously.

holla2040, 2015-01-31:
I think I have a simple workaround to fix server.available(). Need to test it more before I commit it.

spirilis, 2015-02-18:
Any update on this?  Have you played with any reworked WiFiServer code yet?

holla2040, 2015-02-18:
Yes I have. I sent a quick fix to Robert for review a while ago. I'll ping
him and see what he thinks.

davidbuzz, 2015-07-16:
I'd like to see the proposed "quick fix" too, as I've just hit this exact bug.

rajibgupta, 2016-08-17:
Any update on this issue please ?