energia/Energia #560: Reading built-in temperature sensor on Stellaris launchpad

RobisLV, 2015-01-31:
Hi!
I am having problems reading temperature sensor values on my stellaris launchpad (LM4F120H5QR). The values returned by analogRead(TEMPSENSOR) are erratic and do not correlate with actual temperature. I use ubuntu 14.04 and energia 0101E0014. I tried following code:
```
uint32_t TempRead = 0;
float TempC = 0;
void setup()
{
  Serial.begin(9600);
  analogRead(TEMPSENSOR); 
}

void loop()
{
  TempRead = analogRead(TEMPSENSOR);
  TempC = (float)(1475 - ((2475*TempRead)/4096))/10;
  Serial.print("ADC value: ");
  Serial.println(TempRead);
  Serial.print("Temperature in C: ");
  Serial.println(TempC); 
  Serial.println("------------------------");
  delay(1000);
}
```
While runnning this code and putting my finger on MCU this code gives:
```
ADC value: 1939
Temperature in C: 32.40
------------------------
ADC value: 1871
Temperature in C: 34.50
------------------------
ADC value: 1899
Temperature in C: 32.80
------------------------
ADC value: 1895
Temperature in C: 33.00
------------------------
ADC value: 1895
Temperature in C: 33.00
------------------------
ADC value: 1899
Temperature in C: 32.80
------------------------
ADC value: 1899
Temperature in C: 32.80
------------------------
ADC value: 1895
Temperature in C: 33.00
------------------------
ADC value: 1899
Temperature in C: 32.80
------------------------
ADC value: 1899
Temperature in C: 32.80
------------------------
ADC value: 1895
Temperature in C: 33.00
------------------------
ADC value: 1899
Temperature in C: 32.80
```
Basicaly no changes in reading, appart from random-like stuttering. What could be the problem here?

rei-vilo, 2015-01-31:
May I recommend you a better place to post your question and receive more answers from other users?

Try the special threads

* http://forum.43oh.com/forum/28-energia/ at the 43oh forum and 
* http://forum.stellarisiti.com/forum/63-energia/ at the Stellarisiti forum dedicated to the Stellaris LaunchPad.

closed 2015-01-31 by rei-vilo

RobisLV, 2015-01-31:
Ok, I will try to place a post there.