energia/Energia #561: Support for C2000 LaunchPad F2806x

rei-vilo, 2015-02-02:
Some feed-back after having used the **C2000 branch**:

* Energia from the C2000 branch fails with this _catastrophic_ error message:

>
Compilation failure
"/Users/ReiVilo/github/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/Energia.h", line 12: fatal error #1965: cannot open source file "F2806x_Device.h"
1 catastrophic error detected in the compilation of "Blink.cpp".
Compilation terminated.
>

* The folders `F2806x_common` and `F2806x_headers` come with a capital `F` when `f2802x_common` and `f2802x_headers`folders have a lower case `f`.

Please change for `F2802x_common` and `F2802x_headers` with an upper case as it seems to be the standard.

* There a `F2802x_Device.h` but where is `F2806x_Device.h`?

Same for `F2806x_asmfuncs.S`

treygerman, 2015-03-09:
I think this is fixed now.  Will you give it a try Olivier?  Its in both master and c2000.

rei-vilo, 2015-03-09:
No luck! Compiling the _blinky_ example throws an error

>
Compilation failure
"/Users/ReiVilo/github/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/Energia.h", line 12: fatal error #1965: cannot open source file "F2806x_Device.h"
1 catastrophic error detected in the compilation of "Blink.cpp".
Compilation terminated.
>

rei-vilo, 2015-03-11:
Building and linking work for the **LaunchPad F2806x** after commits 15ddc631f3853864e006917d3e79871895c69676 and 7bd28866cc1482254d4fd203d767f7700e5c4706.

The uploader doesn't have the executable permission for Mac OS X.

```
sudo chmod 777 Energia.app/Contents/Resources/Java/hardware/c2000/serial_loader2000/macos/serial_loader2000
```

Next step: uploading.

closed 2015-03-11 by rei-vilo