energia/Energia #564: Wire.cpp #ifdef structure problem

FrankNz, 2015-02-09:
Hi,

Using the simplest 'example' file for a Energia board TivaC TM4C1294 (Evaluation Kit EK-TM4C1924XL) and including Wire.h as an include and in the Makefile. Not making any calls to it I got the following error.

error : /Applications/Energia.app/Contents/Resources/Java/hardware/lm4f/libraries/Wire/Wire.cpp:79:1: Expected declaration before '}' token

The initial pre-processing statements are wrong, as #ifdef __TM4C129XNCZAD__ ... #endif and #ifdef __TM4C1294NCPDT__ ... #endif are independent, resulting in 0, 1 or 2 definitions for  g_uli2cMasterBase. (from a competent person at EmbedxCode Helpdesk)

After some changing the all 'ifdef' blocks in the Wire.cpp file from ...

    static const unsigned long g_uli2cMasterBase[4] =
    {
    #ifdef TARGET_IS_BLIZZARD_RB1
          I2C0_BASE, I2C1_BASE, 
          I2C2_BASE, I2C3_BASE
    #else
    #ifdef __TM4C129XNCZAD__
         I2C0_BASE, I2C1_BASE, 
         I2C2_BASE, I2C3_BASE
    #endif
    #ifdef __TM4C1294NCPDT__
         I2C0_BASE, I2C2_BASE, 
         I2C8_BASE, I2C7_BASE
    #endif
    #endif
    };

 to ...

    static const unsigned long g_uli2cMasterBase[4] =
    {
    #if defined(TARGET_IS_BLIZZARD_RB1)
          I2C0_BASE, I2C1_BASE,
          I2C2_BASE, I2C3_BASE
    #elif defined(__TM4C129XNCZAD__)
          I2C0_BASE, I2C1_BASE,
          I2C2_BASE, I2C3_BASE
    #elif defined(__TM4C1294NCPDT__)
         I2C0_BASE, I2C2_BASE,
         I2C8_BASE, I2C7_BASE
    #else //error
    #error Target for g_uli2cMasterBase NOT defined
    #endif
    };

The problem went away and the code now compiles and you can only have one target!

Being new to this environment I do not know how to submit the updated Wire.cpp file to 'you'.

Regards
Frank

spirilis, 2015-02-18:
Frank- Thanks for reporting this!  I am putting together a "pull request" which I'll associate with this issue to fix this up.  Looks like there were a LOT of places where this needed fixing...

FrankNz, 2015-02-18:
Hi ,

I  think I have succeeded in making a branch of the wire.cpp file against my git login (FrankNZ) and put the changes in the branched file.

( Energia ▸ Contents ▸ Resources ▸ Java ▸ hardware ▸ lm4f ▸ libraries ▸ Wire.cpp)

There seem to be 9 places (block) that are affected.

It may be of use.
Frank

spirilis, 2015-02-18:
No worries, I have the code written, just need to test & verify it with an actual I2C device I have right here.  This also gave me a good reason to pull my poor TM4C1294 launchpad out of its box, had it sitting dormant for a couple months :)  Stand by...

FrankNz, 2015-02-18:
HI Eric,
Thank You, I am looking forward to having a working copy.
Frank