energia/Energia #567: Broadcast UDP packets can't be sent or received when starting Ethernet with fixed IP

solexious, 2015-02-18:
Hello,

While fault finding trying to send broadcast UDP packets from the tiva c connected launchpad I noticed that packets send to the broadcast address for my network (in my case 192.168.1.255) were not reaching any other computers on my network. Likewise any broadcast UDP packets sent out by computers on the network were being received by other computers but not by my tiva c.

Once I switched to obtaining my IP via DHCP when starting Ethernet, both sending and receiving broadcast packets worked.

This is quite an issue for me as my project needs to be able to use both. I've tried looking through the lwip code but haven't been able to come up with the cause or a solution.

solexious, 2015-02-28:
Looking further into this issue it seems to be a subnet issue.

I'm setting only the IP address and letting the Ethernet function fill in the blanks.

Doing this results in this code selecting the subnet mask https://github.com/energia/Energia/blob/master/hardware/lm4f/libraries/Ethernet/Ethernet.cpp#L30

For an IP address of 192.168.1.200 it selects a subnet class of a, but according to this list and my subnet set on computers on this network, it should be a class c http://en.m.wikipedia.org/wiki/IPv4_subnetting_reference

The incorrect subnet leads to an incorrect broadcast address and being unable to send or receive packets from it.

spirilis, 2015-03-01:
that's weird, 192 .... has bit 7 set, so it shouldn't match the class A example.
In fact, 192 has bits 7 and 6 set, so it should fail CLASS_A, CLASS_B and match CLASS_C (192 = 128 + 64).

In fact, here is the bug:
```
const IPAddress CLASS_A_SUBNET(255, 255, 255, 0);
const IPAddress CLASS_B_SUBNET(255, 255, 0, 0);
const IPAddress CLASS_C_SUBNET(255, 0, 0, 0);
```
in Ethernet.h, the subnet definitions are backwards!
Class A = 255.0.0.0, B = 255.255.0.0, C = 255.255.255.0 ....
@robertinant Pull request coming up :-)