energia/Energia #569: Sleep timing issues

Emmanuel-DUFOUR, 2015-02-19:
the "sleep" function behaves differently if the parameter is 999 or below then the time it sleeps is accurate, but if the parameter is 1000 or above (or is using sleepSeconds function ) then it sleeps about 400 ms more than it should. I tested the functions using the code below:
void setup()
{
  Serial.begin(9600);
};

void loop()
{

  sleep(999); //takes about 1001 ms fine
//sleep(1000); //takes about 1400 ms, way off

  Serial.println(millis());

};

rei-vilo, 2015-02-19:
Which MCU please?

Emmanuel-DUFOUR, 2015-02-19:
msp420g2553

spirilis, 2015-02-19:
That's normal.  Above 999 sleepSeconds is used and its precision is off by a bit.

spirilis, 2015-02-19:
I guess the main question then is whether we should be deferring to sleepSeconds at all with large values for sleep.  It does seem like the functions should bear different expectations on precision.  @robertinant what do you think?

rei-vilo, 2015-02-19:
Also keep in mind low power mode.

robertinant, 2015-02-19:
If I remember correctly then sleep() above 999 will call sleep seconds for whole seconds and then sleep for ms left over. Since it is using the Watchdog timer to sleep the increment is 250 ms when calling sleepSeconds() and for the remainder of sleep. So if you call sleep(1001) it will sleep for 1250 ms.

spirilis, 2015-02-19:
@robertinant Yep that is my understanding as well.  Do you think we should change that, so sleep() always performs its sleeps without the help of sleepSeconds?

Emmanuel-DUFOUR, 2015-02-20:
@robertinant it's not off by 250ms but 430ms or so.

spirilis, 2015-02-20:
@Emmanuel-DUFOUR Do you have a 32.768KHz crystal soldered on the board?  If not, it's using the VLOCLK which may account for your 430ms roughly.  Same idea either way.

Emmanuel-DUFOUR, 2015-02-20:
@spirilis I dont have the crystal (I believe, juste using the default launchpad board) and I dont  understand all these clock mechanisms but the code in sleepSeconds says:
******
	if(vlo_freq) {
		SMILLIS_INC = 8192 / (vlo_freq/1000);
		SFRACT_INC = 300;
	} else {
		SMILLIS_INC = 250;
		SFRACT_INC = 0;
	}

	// Activate WDT in ACLK Interval mode
	WDTCTL = WDT_ADLY_250;
*****
In either cases SMILLIS_INC (whatever it means) evaluates to 250. Unless perharps if that "vlo_freq" is defined but not 32.768KHz as it should ?

Emmanuel-DUFOUR, 2015-02-20:
the other thing is that whether we sleep one or 5 seconds the offset stays at 430 ms

Emmanuel-DUFOUR, 2015-02-20:
after some debugging the "vlo_freq" in the code above is defined at 12000

Emmanuel-DUFOUR, 2015-02-20:
That gives a SMILLIS_INC= 682. Here's a blind guess 682-250= 432 which is the offset.

spirilis, 2015-02-20:
Can't remember the exact math & logic behind the additional delay but I bet it's something to do with that.

spirilis, 2015-02-20:
Anyway, if you'd like to play around with sleep() without it using sleepSeconds(), take a look at energia-0101E0014/hardware/msp430/cores/msp430/wiring.c at this part-
```
void sleep(uint32_t milliseconds)
{
        if(milliseconds >= 1000) {
                uint16_t seconds = milliseconds/1000;
                sleepSeconds(seconds);
                milliseconds = milliseconds - (seconds * 1000);
        }
```
Comment out that entire if(milliseconds >= 1000) section.

tapasxplore, 2017-07-17:
The issue is when using sleep() function, the clock source shifts to VLO from DCO to save power. And VLO is not very accurate compared with DCO clock source.

robertinant, 2017-07-18:
VLO has a typical frequency of 12kHz hence this is the value used in the code for vlo_freq. With that said, VLO, as per datasheet, has a minimal value of 4kHz and a maximum value of 20kHz. The frequency it settles on varies by part, temperature, power supply, etc. Hence it can end up anywhere between 4 kHz and 20 kHz. Unfortunately there is no way to measure the VLO Freq. on startup. If you would like accurate sleep times then I suggest that you install the 32kHz watch crystal that came with the LaunchPad in the box. 
Also see this thread for additional information about the sleep() and sleepSeconds() implementation: http://forum.43oh.com/topic/4571-building-low-power-into-energia/?page=2#comment-46495

closed 2017-07-18 by robertinant