energia/Energia #570: Use energia just for build/flash infrastructure

olsonjeffery, 2015-02-21:
Hi,

I'm using Energia with a MSP-EXP430F5529LP on OSX Yosemite. The out-of-the-box experience has been great (I can plug my board into USB, pull any of the examples, build/flash it and it works).

Sadly, I have had zero success getting `mspdebug`, `msp430-gcc`, et al working outside of Energia (installed via Homebrew). I've installed the driver zip and have `/dev/tty.usbmodemxxx` when the board is plugged in, but I can't get mspdebug to connect given various combinations of different driver combinations, etc in the CLI.

From reading [the build process](http://energia.nu/Guide_BuildProcess.html) docs, it seems that the Wiring is actually vanilla C/C++ code with some additional transformations done (like mandatory header includes, the `main` stub that wires-up `setup`, `loop`, etc), disabling the watchdog timer, etc etc.

I'm interested in doing "vanilla", low-level C development outside of the Energia library.

Given my situation/failure to get `mspdebug` working directly on OSX because of the above-outlined issues, is it possible to configure Energia to work in a "no batteries included" mode where it doesn't do all of the mandatory add-ins/transformations before building/flashing?

It'd certainly make a compelling case for an easy-to-use toolchain for the msp430.

If this isn't something the Energia community is interested in, I understand. I just thought I'd ask as I approach different options for getting a low-level dev/build experience set up.

Thanks!

rei-vilo, 2015-02-21:
If you're a Mac user like me, give [embedXcode](http://embedxcode.weebly.com) a try. It is a template for Xcode and uses standard makefiles to build, link, upload and debug. 

For ease of use and convenience, it requires the previous installation of Energia as the app bundles everything: tool-chain, framework, utilities, uploader, etc.

olsonjeffery, 2015-02-21:
Hi @rei-vilo , thanks for the feedback! I went and got it set up and there's definitely a lot to unpack! I'll have to sort through the project infrastructure and see what I can glean from it. Thanks again!

rei-vilo, 2015-02-21:
Enjoy!

closed 2015-02-21 by rei-vilo