energia/Energia #571: Energia does not use -I on sketch's folder during library compile

spirilis, 2015-02-24:
I have been playing with the MSP430 USB Developer's Package (v4.20 I think... whatever is latest as of 2/23/2015) and while I was successful getting a CDC app running under CCS w/ msp430-elf-gcc in Windows, I am having a heck of a time getting it together under Energia.

My intention is to make a "USB" library composed of the USB API and Driverlib, but with the USB Descriptor Tool's output (descriptors.c/.h, and the USB_app reference code for usbEventHandling.c and usbConstructs.c/.h) located inside the user's sketch folder.  This way the user can create custom USB apps using TI's stock Descriptor tool and it will work fine inside Energia.

But I'm running into roadblocks - When Energia goes to build the code inside the "USB" library (within ~/Documents/Energia/libraries/USB), it tries to find descriptors.h - located inside the user's sketch folder - and I would prefer to keep it there, so the USB library doesn't need to house the (ephemeral) descriptors.h file which may vary from sketch to sketch - but can't find it because Energia never bothers to add "-I<path_to_sketch_folder>" when compiling the library code.

Anyway, I think having this feature added -- the "-I<path_to_sketch_folder>" during both sketch-folder file compiles as well as library code compiles would add a dimension of configurability to Energia, as the user would be able to "tune" a library's function by having custom .h files inside their sketch which the library calls upon during its compile.

FWIW, I have seen the USBSerial library that ships with Energia.  It looks like someone really went to town paring down the USB API, and I believe it may have been a version of the USB API pre-driverlib dependent, but most importantly, the descriptors.c/.h is housed inside the USBSerial library, which makes it much less "general-purpose" for others' intended USB applications.

StefanSch, 2015-03-03:
Hi, 
i have ported the initial version of the USBSerial library.
On the initial porting i have seen the some issue, therefore i have generated the USBSerial library instead of a global USBLib.
A few days ago i also updated the version to the latest USB driver but did not publish it so far.

I agree it would be great if the Energia compile system could be extended to include also the sketch folder for the build of the libraries. Another option would be to pass the data from the sketches to the driver via pointers but esp. for the USB this could be difficulty as this also has a lot of type declarations and definitions inside the files generated by the descriptor tools and also some output files of the descriptor tools needs to be adopted for the Energia system.
I will have a closer look into this over the next days maybe i get a great idea on how to solve this easily or discuss this with the Energia main developer.

spirilis, 2015-03-03:
@StefanSch Thanks for chiming in... yeah I dropped my work on it after submitting this issue, looked like quite a mess.

spirilis, 2016-02-11:
Closing this for now - it will probably have to wait until @robertinant upgrades to an Arduino 1.6-based codebase.

closed 2016-02-11 by spirilis