energia/Energia #575: C2000 LaunchPad F2806x — Building and Linking on Mac OS X

rei-vilo, 2015-03-12:
I'm sorry to re-open the case #561 for Mac OS X. 

When I build Energia with `ant dist` and run it on the `~/github/Energia/Energia/build/macosx/work/` folder, all goes fine.

* If I **copy** Energia and paste it into the `/Applications` folder, I get this error.

[image]

* If I **move** Energia from `~/github/Energia/Energia/build/macosx/work/`into the `/Applications` folder, compilation fails.

>
"/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/Energia.h", line 12: fatal error #1965: cannot open source file "F2806x_Device.h"
1 catastrophic error detected in the compilation of "Blink.cpp".
Compilation terminated.
>

This is generally caused by a hard-coded path.

rei-vilo, 2015-03-13:
For the **F28027**, 

* `F2802x_Device.h` is located at `Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/` 
* and `F2802x_Device.h` includes
```
#include "f2802x_headers/include/F2802x_Adc.h"                // ADC Registers
```

For the **F29069**, 
* `F2806x_Device.h` is located at `Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/F2806x_headers/include` 
* and `F2806x_Device.h` includes
```
#include "F2806x_Adc.h"                // ADC Registers
```
but the path to `Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/F2806x_headers/include` isn't mentioned among the parameters passed on to `Energia.app/Contents/Resources/Java/hardware/tools/c2000/bin/cl2000`.

Is there a reason for having two different logics?

treygerman, 2015-03-13:
They're different because I tried to improve the structure of the F2802x files when I originally released that stuff.  I think the issue should be fixed now.

rei-vilo, 2015-03-13:
File `/Users/ReiVilo/github/Energia/Energia/hardware/c2000/cores/c2000/F2806x_headers/include/F2806x_Device.h` has been removed by commit 014c02ea2c7312a1416acd8c79ca10fe61613e26 but it is still asked for by `cl2000`.

treygerman, 2015-03-13:
Which file is including it?  This builds fine on my mac :-/

rei-vilo, 2015-03-13:
Energia.h includes F2806x_Device.h, which is no longer available. Last commit deleted it, see https://github.com/energia/Energia/commit/014c02ea2c7312a1416acd8c79ca10fe61613e26

rei-vilo, 2015-03-13:
The `ant clean` and `ant dist` procedure goes fine and build Energia.

However, compilation with Energia fails, as the F2806x_Device.h is no longer available.

See deleted file at https://github.com/energia/Energia/commit/014c02ea2c7312a1416acd8c79ca10fe61613e26

rei-vilo, 2015-03-13:
My guess is the `PATH` system variable on your Mac includes the folder where the `F2806x_Device.h` file is located.

treygerman, 2015-03-16:
What include paths show up for you when you try to compile?

I checked my path and there is nothing in there related to any Energia directories.  

On my machine Energia.h is including F2806x_Device.h.  F2806x_Device.h is in hardware/c2000/cores/c2000 which is one of the include paths passed to the compiler.

rei-vilo, 2015-03-16:
The file `hardware/c2000/cores/c2000/F2806x_headers/include/F2806x_Device.h` can't be included in your configuration as it has been removed by commit 014c02ea2c7312a1416acd8c79ca10fe61613e26

[image]

rei-vilo, 2015-03-16:
If I add the file `F2806x_Device.h` manually, then another error is raised:

>
"/Applications/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/F2806x_Device.h", line 173: fatal error #1965: cannot open source file "F2806x_Adc.h"
1 catastrophic error detected in the compilation of "/Applications/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/F2806x_common/source/F2806x_Adc.c".
Compilation terminated.
>

rei-vilo, 2015-03-19:
I'll wait for the official release featuring the F28069.

rei-vilo, 2015-03-25:
Compilation for C200 F28069 on Energia 0101E0015 release fails

>
Compilation failure
"/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000/Energia.h", line 12: fatal error #1965: cannot open source file "F2806x_Device.h"
1 catastrophic error detected in the compilation of "Blink.cpp".
Compilation terminated.
>

The file `F2806x_Device.h` is missing from the distribution.

ozsavran, 2015-04-07:
Hello Trey,

Utilizing your work from the other side of the globe, in a mid-size Mediterranean city Mersin, Turkey.  I am an organic farmer (and retired engineer) trying to build 3-phase controllers for our pumps.  The LaunchXL-28027F and -28069M boards show different behaviour on different Windows computers, unfortunately.  Maybe its due to Windows versions, maybe due to different USB controllers on motherboards, dunno.

Both cards open 2 USB XDS100 channels.  When VCP is allowed and USB connection is restarted with a quick disable/enable, they show up as COM3/COM4 etc.  You can change them to COM1/COM2, no problem.  Speed of COM port does not have an effect as long as setup is 8/1/0.  Putty on COM2 (XDS Channel B) only works, at 115200bps setting in Putty.

Now, to stick to the subject in this thread here, I get the very same message about not finding  "F2806x_Device.h" like rei-vilo quoted above, on all Windows-7/32-bit laptop, Windows Vista-64 desktop, and Windows-8 tablet PC.

The reason I try them all is, although Windows-7/32 successfully programs Blink & Fade programs on the F28027F card,  Windows Vista-64 can not for some reason, although COM ports show up fine. Tried latest FTDI drivers, etc.  But that's a seperate issue.  Question on this thread is, how do we compile something onto the F28069M board that we can actually see on all Windows versions, due to lack of C libraries please?

Thanks for taking the time for us. :)

rei-vilo, 2015-04-17:
@treygerman

Still some issues at linking with today's commit edc070fc9a172af7d4d574e9313c50dcde954673.

>
/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/tools/c2000/bin/cl2000 -v28 -ml -mt -g --gcc --define=ENERGIA=15 --define=F_CPU=90000000L --define=TMS320F28069 --define=ARDUINO=101 --diag_warning=225 --display_error_number --diag_wrap=off -z --stack_size=0x300 --warn_sections -i"/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000" -i"/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/include" -i"/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/variants/launchpad_f28069" -i"/Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/../tools/c2000/lib" --reread_libs --display_error_number --diag_wrap=off --entry_point=code_start --rom_model -o/var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build5377813721124357647.tmp/Blink.cpp.out /var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build5377813721124357647.tmp/Blink.cpp.o /var/folders/5d/dky9xc691m32kztqnhkn8bhw0000gn/T/build5377813721124357647.tmp/core.a -lrts2800_ml.lib /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000//F2806x_common//cmd//F28069.cmd /Users/ReiVilo/GitHub/Energia/Energia/build/macosx/work/Energia.app/Contents/Resources/Java/hardware/c2000/cores/c2000//F2806x_headers//cmd//F2806x_Headers_nonBIOS.cmd 
<Linking>
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<wiring.c.o>"
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<wiring_analog.c.o>"
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<wiring_digital.c.o>"
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<main.cpp.o>"
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<HardwareSerial.cpp.o>"
error #10056: symbol "_pin_mapping" redefined: first defined in "Blink.cpp.o"; redefined in "core.a<Print.cpp.o>"
warning #10210-D: creating ".esysmem" section with default size of 0x400; use the -heap option to change the default size
warning #10063-D: entry-point symbol other than "_c_int00" specified:  "code_start"
error #10010: errors encountered during linking; "Blink.cpp.out" not built
>
Compilation failure
>

treygerman, 2015-04-17:
Yup...on my list of things to fix.  The reason I wasn't seeing the F2806x compile fail was because I had a untracked file which fixed the issue...

treygerman, 2015-05-04:
Should be fixed now

rei-vilo, 2015-05-04:
Indeed, building and linking work for the F28069. Thank you, @treygerman!

rei-vilo, 2015-05-04:
Even if the feature isn't available on today's release 15 of Energia, I close the topic as it is solved. 

The fix will be included in next release 16.

closed 2015-05-04 by rei-vilo