energia/Energia #576: EEPROM Library Not working, crashes.  LM4C, Connected Launchpad

ekerin, 2015-03-22:
Using the connected launchpad (TM4C1294XL) - and the EEPROM library, crashes the app (Same code works on Arduino)

I found this note in the Silicon Errata: Mem#11 on Page 51 (http://www.ti.com/lit/er/spmz850d/spmz850d.pdf)

The ROM_EEPROMInit API in TivaWare does not correctly initialize the EEPROM
module as described in the data sheet. It should not be used to initialize the EEPROM.
Workaround(s): Use the Flash version of the EEPROMInit API in TivaWare version 2.1 or later.

The following patch will reconfigure the map for the connected launchpad, and switch to using the MAP version of the EEPROM write code in EEPROM.cpp, instead of the hard coded ROM version.  Only tested on Connected Launchpad (TM4C1294XL).

--- rom.h.orig  2014-07-19 19:11:54.428743038 -0400
+++ rom.h       2014-07-19 19:12:32.231312244 -0400
@@ -889,8 +889,7 @@
 //
 //*****************************************************************************
 #if defined(TARGET_IS_BLIZZARD_RA3) ||                                        \
-    defined(TARGET_IS_BLIZZARD_RB1) ||                                        \
-    defined(TARGET_IS_SNOWFLAKE_RA0)
+    defined(TARGET_IS_BLIZZARD_RB1) 
 #define ROM_EEPROMRead                                                        \
         ((void (*)(uint32_t *pui32Data,                                       \
                    uint32_t ui32Address,                                      \
@@ -973,8 +972,7 @@
         ((uint32_t (*)(void))ROM_EEPROMTABLE[12])
 #endif
 #if defined(TARGET_IS_BLIZZARD_RA3) ||                                        \
-    defined(TARGET_IS_BLIZZARD_RB1) ||                                        \
-    defined(TARGET_IS_SNOWFLAKE_RA0)
+    defined(TARGET_IS_BLIZZARD_RB1)
 #define ROM_EEPROMProgram                                                     \
         ((uint32_t (*)(uint32_t *pui32Data,                                   \
                        uint32_t ui32Address,                                  \
--- EEPROM.cpp.orig     2015-03-22 13:25:28.832987269 -0400
+++ EEPROM.cpp  2014-07-19 19:16:24.228934315 -0400
@@ -48,7 +48,7 @@
        //int word = (address / BYTES_PER_WORD) % WORDS_PER_BLOCK;
        unsigned long wordVal = 0;
 
-       ROM_EEPROMRead(&wordVal, byteAddr, 4);
+       MAP_EEPROMRead(&wordVal, byteAddr, 4);
        wordVal = wordVal >> (8*(address % BYTES_PER_WORD));
 
        return (uint8_t) wordVal;
@@ -59,11 +59,11 @@
        unsigned long byteAddr = address - (address % BYTES_PER_WORD);
 
        unsigned long wordVal = 0;
-       ROM_EEPROMRead(&wordVal, byteAddr, 4);
+       MAP_EEPROMRead(&wordVal, byteAddr, 4);
        wordVal &= ~(0xFF << (8*(address % BYTES_PER_WORD)));
        wordVal += value << (8*(address % BYTES_PER_WORD));
 
-       ROM_EEPROMProgram(&wordVal, byteAddr, 4);
+       MAP_EEPROMProgram(&wordVal, byteAddr, 4);
 }
 
 EEPROMClass EEPROM;

robertinant, 2015-03-23:
Which EEPROM library is this?

ekerin, 2015-03-23:
The one in Energia 14 (also happens in 11 and 12)

robertinant, 2017-08-30:
This issue was moved to energia/tivac-core#20

closed 2017-08-30 by robertinant