energia/Energia #577: Add SPI CS pins for "Wireless" "Display" "Other" to line up with BoosterPack Standard

measley, 2015-03-23:
need to add support to SPI library for additional CS pins as designated by the BoosterPack standard found at www.ti.com/byob 

SPI CS Wireless - pin 18
SPI CS Display - pin 13
SPI CS Other - pin 12

spirilis, 2015-03-24:
I'm not sure what this is trying to accomplish.  Energia doesn't treat the output-side (master driven) SPI pins special in any way; the pins are specified typically to a library and the library uses digitalWrite to drive those pins as needed.  The SPI library itself does not have intrinsic support for driving the SPI Chip Select pins automatically.  (Actually I guess the CC3200 has some funky stuff for that .... but I don't think that's the standard Arduino way to do it).

spirilis, 2015-03-24:
Well ... hmm, I guess adding the pin #'s with a symbolic name to the variants pins_energia.h is what you might be trying to accomplish.  In that case yeah, something could be added...

robertinant, 2016-08-29:
I am not going to add symbolic names for those pins. They should be documented on the BoosterPack pin maps / schematic.

closed 2016-08-29 by robertinant