energia/Energia #578: Energia MT: Wrong Link for Mac OS X

rei-vilo, 2015-03-25:
The link to download Energia MT for Mac OS X on http://energia.nu/download/ points to previous release 14.

> Download here: energia-0101E0015-macosx.dmg 

points to 

`http://energia.nu/downloads/downloadv3.php?file=energia-0101E0014-macosx.dmg`

robertinant, 2015-03-25:
Thanks for reporting! fixed..

rei-vilo, 2015-03-25:
Still 

>
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>F9276DED4D07EB9E</RequestId>
<HostId>
CaT5QjqD6E4Xdka4RQZHwYVj1a+8oy9S0vshzzTkqQGNCL0hYqUUHmj/3W2cHL5T
</HostId>
</Error>
>

robertinant, 2015-03-25:
Fixed. Issue on Amazon AWS not making the file public.

closed 2015-03-25 by robertinant