energia/Energia #579: No font hinting / antialiasing in Linux / Ubuntu

anton-veretenenko, 2015-03-25:
There is no font hinting / antialising in Linux / Ubuntu version. It does not respect system settings.
Two screenshots attached, using the same font face.
This is how it looks in Energia without hinting.
[image]

This is how it should look when hinted (do not mind colors).
[image]

rei-vilo, 2015-03-25:
Have a look at the file `preferences.txt` and set editor.antialias` to `true`.

```
editor.antialias=true
```

On Mac OS X, the file is located under `~/Library/Energia/preferences.txt`.

anton-veretenenko, 2015-03-25:
yep it works, close it then :)

closed 2015-03-25 by anton-veretenenko