energia/Energia #580: usb_hid_feat

a-bhat1, 2015-03-26:
Add USB HID library and example to Energia.   The included example sends to and receives echoed data from the micro-controller.

spirilis, 2015-03-26:
We've tried this before.  It's not simple.  I gave it a go about a month ago... Energia's build commands make it tricky to implement the MSP430 USB Developer's Package and Descriptor Tool outputs in a manner that's flexible enough for sketches to use.

rei-vilo, 2015-03-28:
Which board?

a-bhat1, 2015-03-31:
USBHID library has been tested on the MSP430F5529 board.  The USB stack has the ability to work with the F5xx and F6xx family of boards, however, I am not sure what needs to happen in Energia for HID library to work with boards other than MSP430F5529.

robertinant, 2016-08-29:
This issue was moved to energia/msp430-lg-core#19

closed 2016-08-29 by robertinant