energia/Energia #583: MSP432: energia produces bad .elf, huge .bin files.

WestfW, 2015-03-28:
Energia 0101E0015 produces huge .bin files (500MB+) for trivial examples on MSP432.
This turns out to be because it's producing a .elf file with initialized data in RAM (at 0x20000000) as well as code at 0x0 (this example from Blink.ino)

`
0000915c R ti_sysbios_family_arm_m3_Hwi_E_hardFault__C
00009160 R ti_sysbios_knl_Semaphore_Object__DESC__C
00009180 R ti_sysbios_knl_Idle_funcList__C
00009188 R xdc_runtime_Error_E_memory__C
0000918c R ti_runtime_heaps_HeapRem_Module__FXNS__C
000091b4 R ti_sysbios_family_arm_m3_Hwi_E_noIsr__C
000091b8 R xdc_runtime_Text_charCnt__C
000091bc R ti_sysbios_family_arm_m3_Hwi_E_busFault__C
000091c0 R ti_sysbios_knl_Swi_Object__count__C
000091c4 R ti_runtime_heaps_HeapRem_Object__PARAMS__C
000091dc A __data_load__
00009db8 R __exidx_start
00009dc0 N __TRDR__
00009dc0 R __exidx_end
00009de8 N __TARG__
00009e0c N __PLAT__
00009e34 N __ISA__
00009e48 N __ASM__
00009ebc N __UNUSED_FLASH_start__
00010000 a __SRAM_LENGTH__
00040000 a __FLASH_LENGTH__
00040000 a __UNUSED_FLASH_end__
20000000 a _vtable_base_address
20000000 D ti_sysbios_family_arm_m3_Hwi_ramVectors
20000140 D __data_start__
20000140 D func_ptr
20000148 D taskNames
2000014c D ti_sysbios_family_arm_m3_Timer_Object__table__V
2000017c D ti_sysbios_knl_Swi_Object__table__V
200001ac D ti_sysbios_knl_Swi_Module__state__V
200001c8 D ti_sysbios_knl_Semaphore_Object__table__V
200001dc D ti_sysbios_family_arm_m3_Timer_Module__root__V
200001e4 D ti_sysbios_knl_Clock_Module__root__V
200001ec D ti_sysbios_heaps_HeapMem_Module__root__V
200001f4 D ti_sysbios_BIOS_Module__state__V
20000218 D xdc_runtime_SysMin_Module__state__V
20000224 D ti_sysbios_knl_Semaphore_Module__root__V
2000022c D ti_sysbios_family_arm_msp432_Timer_Module_State_0_handles__A
2000023c D ti_sysbios_gates_GateHwi_Module__root__V
`

This sounds like a linker script issue: initialized variables should appear in the flash address space in the finished .elf file, and get copied to RAM by startup code.   This also implies that the binary won't work if burnt to flash...

rei-vilo, 2015-03-28:
Although the .elf file is 500+ KB, the actual size of the sketch is much smaller.

> Binary sketch size: 45763 bytes (of a 262144 byte maximum)

Use `arm-none-eabi-size` to learn more

WestfW, 2015-03-28:
The .elf file is of reasonable size.   It's the .bin file that is huge.
It's the .bin file that gets uploaded, right?  Does the uploader automatically skip large regions of zeros?
(and there's still the "how can it run from flash if RAM variables are initialized via the upload"?)

rei-vilo, 2015-03-28:
No, the `.bin` file isn't uploaded.

WestfW, 2015-03-28:
I don't have an actual board yet, but the "upload" command produces:
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/bin/lm4flash /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/build6157934834490470798.tmp/Blink.cpp.bin 
    Unable to find any ICDI devices

Note: not 512kbytes.  512 **M**bytes!

spirilis, 2015-03-28:
What is the .bin file for if it's not used to upload?

rei-vilo, 2015-03-29:
On Mac OS X, `.elf` is uploaded, not `.bin`.

`DSLite -c MSP432P401R.ccxml -f MultiBlink.cpp.elf`

`.bin` is 536 MB but `.elf` is 1.3 MB only.

```
total 1053064
drwxr-xr-x   10 staff        340 29 mar 08:16 .
drwx------  101 staff       3434 29 mar 08:18 ..
-rw-r--r--    1 staff       2465 29 mar 08:16 MultiBlink.cpp
-rwxr-xr-x    1 staff  536879240 29 mar 08:16 MultiBlink.cpp.bin
-rwxr-xr-x    1 staff    1272230 29 mar 08:16 MultiBlink.cpp.elf
-rw-r--r--    1 staff     953870 29 mar 08:16 MultiBlink.cpp.map
-rw-r--r--    1 staff      17864 29 mar 08:16 MultiBlink.obj
-rw-r--r--    1 staff        605 29 mar 08:16 Variables.mk
-rw-r--r--    1 staff       4068 29 mar 08:16 main.cpp
-rw-r--r--    1 staff      20500 29 mar 08:16 main.obj
```

robertinant, 2015-03-29:
Interesting find! I did use the .bin file in the early on but then changed it to uploading the elf file. I will remove the bin file generation from the Makefile. Will still look into why the elf file is so huge.

rei-vilo, 2015-03-29:
However, `size` reports a very acceptable size for the basic multi-blinky sketch, and a highly optimised size for the RTOS core.

```
$ arm-none-eabi-size MultiBlink.cpp.elf 
   text	   data	    bss	    dec	    hex	filename
  37447	   4404	   3912	  45763	   b2c3	MultiBlink.cpp.elf
```

WestfW, 2015-03-29:
It looks like it is probably the .vtable section that has an incorrect LMA (in a linker map somewhere?)  Otherwise, the .data segment symbols in the .elf file seem identical to other targets (symbol values are 20000000+, but LMA for .data is down in low flash.)   (It's a little hard to tell: I can't find an example that produces .vtable for other targets.)  (Hmm.  Perhaps .bss is wrong as well (but I'm not sure that matters.)  In the `objdump -h` output belong, .bss and .vtable are the only segments with LMAs up in the RAM segment.)

~~~~

Blink.cpp.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .intvecs      0000003c  00000000  00000000  00008000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .vtable       00000140  20000000  20000000  00020000  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 .text         000096d8  00000040  00000040  00008040  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .c_int00      00000044  00009718  00009718  00011718  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .rodata       00000b3c  00009760  00009760  00011760  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .rodata_ti_sysbios_family_arm_m3_Hwi_E_reserved__C 00000004  0000a29c  0000a29c  0001229c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  6 .rodata_ti_sysbios_knl_Task_Object__heap__C 00000004  0000a2a0  0000a2a0  000122a0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  7 .rodata_xdc_runtime_Startup_maxPasses__C 00000004  0000a2a4  0000a2a4  000122a4  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  8 .rodata_xdc_runtime_Startup_sfxnRts__A 0000001c  0000a2a8  0000a2a8  000122a8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  9 .rodata_ti_sysbios_gates_GateMutex_Module__FXNS__C 00000024  0000a2c4  0000a2c4  000122c4  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 10 .rodata_ti_sysbios_knl_Task_numConstructedTasks__C 00000004  0000a2e8  0000a2e8  000122e8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 11 .rodata_ti_sysbios_knl_Task_Object__PARAMS__C 0000003c  0000a2ec  0000a2ec  000122ec  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 12 .rodata_xdc_runtime_Core_Module__diagsMask__C 00000004  0000a328  0000a328  00012328  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 13 .rodata_xdc_runtime_Error_policyFxn__C 00000004  0000a32c  0000a32c  0001232c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 14 .rodata_xdc_runtime_IGateProvider_Interface__BASE__C 00000004  0000a330  0000a330  00012330  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 15 .rodata_ti_sysbios_knl_Task_Object__count__C 00000004  0000a334  0000a334  00012334  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 16 .rodata_ti_sysbios_knl_Task_defaultStackSize__C 00000004  0000a338  0000a338  00012338  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 17 .rodata_ti_sysbios_family_arm_lm4_Timer_startupNeeded__C 00000004  0000a33c  0000a33c  0001233c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 18 .rodata_ti_sysbios_family_arm_lm4_Timer_enableFunc__C 00000004  0000a340  0000a340  00012340  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 19 .rodata_xdc_runtime_Error_policy__C 00000001  0000a344  0000a344  00012344  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 20 .rodata_xdc_runtime_Text_charTab__A 00000004  0000a348  0000a348  00012348  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 21 .rodata_ti_sysbios_family_arm_m3_Hwi_excHookFuncs__C 00000004  0000a34c  0000a34c  0001234c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 22 .rodata_ti_sysbios_knl_Task_allBlockedFunc__C 00000004  0000a350  0000a350  00012350  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 23 .rodata_xdc_runtime_IHeap_Interface__BASE__C 00000004  0000a354  0000a354  00012354  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 24 .rodata_ti_sysbios_knl_Task_hooks__C 00000008  0000a358  0000a358  00012358  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 25 .rodata_xdc_runtime_Startup_firstFxns__A 00000010  0000a360  0000a360  00012360  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 26 .rodata_ti_sysbios_knl_Clock_Module_State_clockQ__O 00000004  0000a370  0000a370  00012370  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 27 .rodata_xdc_runtime_Error_maxDepth__C 00000002  0000a374  0000a374  00012374  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 28 .rodata_ti_sysbios_utils_Load_autoAddTasks__C 00000002  0000a376  0000a376  00012376  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 29 .rodata_ti_sysbios_family_arm_m3_Hwi_E_debugMon__C 00000004  0000a378  0000a378  00012378  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 30 .rodata_ti_sysbios_knl_Task_initStackFlag__C 00000002  0000a37c  0000a37c  0001237c  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
[... 244 lines cut ...]
153 .debug_info   00079d31  00000000  00000000  00022664  2**0
                  CONTENTS, READONLY, DEBUGGING
154 .debug_abbrev 0000d75b  00000000  00000000  0009c395  2**0
                  CONTENTS, READONLY, DEBUGGING
155 .debug_aranges 000030c0  00000000  00000000  000a9af0  2**3
[... cut ...]
160 .ARM.attributes 00000039  00000000  00000000  00107c41  2**0
                  CONTENTS, READONLY
161 .debug_frame  00008d10  00000000  00000000  00107c7c  2**2
                  CONTENTS, READONLY, DEBUGGING
162 .debug_loc    00029f5d  00000000  00000000  0011098c  2**0
                  CONTENTS, READONLY, DEBUGGING
163 .stab         00000054  00000000  00000000  0013a8ec  2**2
                  CONTENTS, READONLY, DEBUGGING
164 .stabstr      000000ac  00000000  00000000  0013a940  2**0
                  CONTENTS, READONLY, DEBUGGING
~~~~

robertinant, 2016-08-29:
The bin file is no longer generated for msp432. The .elf file us used to flash the target using DSLite.

closed 2016-08-29 by robertinant