energia/Energia #585: MSP432: modified core files are not recompiled.

WestfW, 2015-03-28:
If I modify one of the core libraries for MSP432 (ie .../hardware/msp432/cores/msp432/wiring.c), a sketch build that should be using that file does not recompile the modified version.

WestfW, 2015-03-28:
Hmm.  I see that "preferences" has a "compile driverlib from source" option now, that I guess is supposed to address this?  But it doesn't seem to work:

    armlink Blink.obj main.obj
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/bin/arm-none-eabi-gcc -Os @"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/compiler.opt" -nostartfiles -Wl,--no-wchar-size-warning -Wl,-static -Wl,--gc-sections -L"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common" -L"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/lib"  Blink.obj  main.obj    -Wl,-T"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/linker.cmd" -L/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R -L/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/libs   -lstdc++ -lgcc -lc -lm -lnosys -Wl,-Map=Blink.cpp.map -o Blink.cpp.elf
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/runtime/wiring/msp432/lib/ti.runtime.wiring.msp432.lib(wiring_analog.om4fg): In function `analogWrite':
    /db/ztree/library/trees/emt/emt-a28/src/ti/runtime/wiring/msp432/wiring_analog.c:259: undefined reference to `GPIO_setAsPeripheralModuleFunctionOutputPin'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/runtime/wiring/msp432/lib/ti.runtime.wiring.msp432.lib(wiring_analog.om4fg): In function `stopAnalogRead':
    /db/ztree/library/trees/emt/emt-a28/src/ti/runtime/wiring/msp432/wiring_analog.c:561: undefined reference to `GPIO_setAsPeripheralModuleFunctionOutputPin'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/drivers/lib/drivers_MSP432P401R.am4fg(GPIOMSP432.om4fg): In function `GPIO_hwiIntFxn':
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:244: undefined reference to `GPIO_getEnabledInterruptStatus'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:247: undefined reference to `GPIO_clearInterruptFlag'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/drivers/lib/drivers_MSP432P401R.am4fg(GPIOMSP432.om4fg): In function `GPIO_setConfig':
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:446: undefined reference to `GPIO_interruptEdgeSelect'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:448: undefined reference to `GPIO_clearInterruptFlag'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:386: undefined reference to `GPIO_setAsInputPinWithPullDownResistor'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:383: undefined reference to `GPIO_setAsInputPinWithPullUpResistor'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:394: undefined reference to `GPIO_setAsOutputPin'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:401: undefined reference to `GPIO_setDriveStrengthHigh'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:404: undefined reference to `GPIO_setOutputHighOnPin'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:389: undefined reference to `GPIO_setAsInputPin'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:397: undefined reference to `GPIO_setDriveStrengthLow'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:407: undefined reference to `GPIO_setOutputLowOnPin'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/drivers/lib/drivers_MSP432P401R.am4fg(GPIOMSP432.om4fg): In function `GPIO_write':
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:505: undefined reference to `GPIO_setOutputLowOnPin'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/gpio/GPIOMSP432.c:499: undefined reference to `GPIO_setOutputHighOnPin'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/drivers/lib/drivers_MSP432P401R.am4fg(PWMTimerMSP432.om4fg): In function `PWMTimerMSP432_close':
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/pwm/PWMTimerMSP432.c:109: undefined reference to `Timer_A_setCompareValue'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/pwm/PWMTimerMSP432.c:110: undefined reference to `Timer_A_stopTimer'
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/drivers/lib/drivers_MSP432P401R.am4fg(PWMTimerMSP432.om4fg): In function `PWMTimerMSP432_control':
[... cut ...]
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/power/PowerMSP432_tirtos.c:110: undefined reference to `CPU_cpsie'
    /db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/power/PowerMSP432_tirtos.c:110: undefined reference to `CPU_cpsie'
    collect2: error: ld returned 1 exit status
    make: *** [Blink.cpp.elf] Error 1
    /Applications/arduino/Energia-e15.app/Contents/Resources/Java/tools/common/bin/make returned 2

robertinant, 2015-03-29:
Thanks for the report @WestfW really appreciate it! 

1: Compile from source. This will be exposed in the GUI in the next version. For now there is an option that you can set to yes in hardware/common/Makefile called CORE_FROM_SOURCE. Currently this is set to no, change it to yes and you will compile from source.

2: You stumbled upon a bug with the "compile driverlib from source" option for MSP432. Would you please be so kind to open a separate issue for this so that we can keep track of it in it's own issue?

cevatbostancioglu, 2015-08-09:
i think he try to edit msp432 energia firmware. 
energia msp432 firmware have a 2 serial ports and msp432 have a 4 hw serial ports. so i edit the board_init.c file. but doesn't work. so i edit board_init.c with syntax error. so i see core libraries are not compiled. 
last two days i try to run makefile with "make -f makefile" bla bla but they didn't work too.
so i found this topic. thank you so much. i will try CORE_FROM_SOURCE. if it's works or not , i will text here.

cevatbostancioglu, 2015-08-09:
hello again , 
i edit CORE_FROM_SOURCE=yes , and i try to compile.
there is an errors and solutions.
1.)wiring_analog.c -> compiler can't find rom.h  , i edit <driverlib/rom.h>
2.)common\ti\dirver\power , in .c and .h files cannot find driverlib too. so driverlib is far from there. so i copied pcm.c , pcm.h , cs.c , cs.h to the folder. and edit "pcm.h" , "cs.h" like that. thats okey now.
3.wiring.c cannot find driver lib. so i solve like solution 1.

but there is a still problem there. i edit board_init.c with syntax error(i delete '{' in function) , but there is no error on energia ide. so energia ide still don't compile board_init.c ?

cactrot, 2015-09-12:
You are correct it doesn't compile it. I am still learning the ins and outs of recursive makefiles so I used hard coded paths.

C:\energia\hardware\msp432\variants\MSP_EXP432P401R\Board_init.c
line: 48 to 56
```
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/rom.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/rom_map.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/dma.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/gpio.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/i2c.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/spi.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/timer_a.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/uart.h>
#include <C:\energia\hardware\msp432\cores\msp432\driverlib/wdt_a.h>
```

line: 690
```
{57600,  12000000,  13,  0,  37, 1},
{57600,   6000000,   6,  8,  32, 1},
{57600,   3000000,   4,  3,  2,  1},
```

C:\energia\hardware\msp432\variants\MSP_EXP432P401R\makedefs
line: 70
```
PREFIX=C:\energia\hardware\tools\lm4f\bin\arm-none-eabi
```
I removed the -p from the mkdir command since windows doesn't like that.

C:\energia\hardware\msp432\variants\MSP_EXP432P401R\Makefile
line: 70
```
	@mkdir  lib/${COMPILER}-cm4f
```

Then ran:
C:\energia\hardware\msp432\variants\MSP_EXP432P401R>C:\energia\tools\common\bin\
make.exe

The output was named: emt432LP.m4fg.lib

Energia was looking for a lib named: Board.m4fg.lib

So I renamed the output to that and finally got standard firmata compiled with 57600 baud serial.

Hope I haven't forgotten any steps and that someone find this useful.

cevatbostancioglu, 2015-09-15:
hello cactrot.
i didn't understand Then ran:
C:\energia\hardware\msp432\variants\MSP_EXP432P401R>C:\energia\tools\common\bin\
make.exe step.
can you explain again?

cevatbostancioglu, 2015-09-23:
can you explain @cactrot ?

cactrot, 2015-09-23:
It appears I did forget a step I had to edit "makedefs" file to remove the "if" statements that would be used to print progress info.

http://pastebin.com/p8GWTpsE

cevatbostancioglu, 2015-09-23:
hello @cactrot thank for reply.
i can't still can't compile the makefile.
there is a picture from command prompt.
[image]

 i try my win8.1 machine but make.exe cannot run on win8.1 64 bit. actually first 5 minutes it's works and same output like picture. but then not working. 
anyway , i want to change board_init.c and recompile it like you. can you explain like you did up again? this is very important for me because i am working on like more than 1 month and you are the only person did this i know.

cevatbostancioglu, 2015-09-23:
i write make -p C:\Documents and Settings\cevat\Desktop\energia0101E0016\hardware\msp432\variants\MSP_EXP432P401R 
but it's  not create the lib files like you.

closed 2016-08-29 by robertinant