energia/Energia #586: MSP432: Windows makefile doesn't allow for

WestfW, 2015-03-28:


rei-vilo, 2015-03-28:
The requirement for the names and paths without spaces or special characters, is a limitation due to the GCC tool-chain.

WestfW, 2015-03-28:
Note that targets other than MSP432 work fine with an install in "\Program Files", even though they have both source and tools there.  It's only the (new) Make-based build that has problems.

rei-vilo, 2015-03-28:
That's a limitation of the `makefile`.

WestfW, 2015-03-29:
If you can figure out how to put a "dospath.bat" into a well know executable place (even if the current IDE location has spaces in the dirs), then you can convert the paths to their DOS equivalents in the Makefile...

dospath.bat:

    @ECHO OFF
    echo %~s1

Changed part of hardware/common/Makefile:

    # (if present) include definitions of the macros described above

    include Variables.mk

    ifeq ($(OS),Windows_NT)
    $(shell set CYGWIN = nodosfilewarning)
    RM = $(shell rmdir /S /Q build >nul 2>nul)
    MKDIR = mkdir
    SHELL = C:/Windows/System32/cmd.exe
    APPLICATION_PATH := $(shell "c:\bin\dospath" "$(APPLICATION_PATH)")
    CLOSURE := $(shell "c:\bin\dospath" "$(CLOSURE)")
    SDKROOT := $(shell "c:\bin\dospath" "$(SDKROOT)")
    CCROOT := $(shell "c:\bin\dospath" "$(CCROOT)")
    else
    RM = rm -rf build
    MKDIR = mkdir -p
    endif

    $(info TREEROOT $(TREE_ROOT))
    $(info APPLICATION_PATH $(APPLICATION_PATH))

    BOARD_DIR := $(APPLICATION_PATH)/hardware/$(PLATFORM)/targets/$(BOARD)
    include $(BOARD_DIR)/Board.mk

    $(info CCROOT $(CCROOT))
    $(info SDKROOT $(SDKROOT))

WestfW, 2015-03-29:
You can apparently do this without a .bat file by calling cmd.exe (shell) with an argument like:

    cmd /c 'for %I in ("\Program Files\Energia-0101E0015\tools\common\") do @echo %~sI'

But the quoting of quotes and parens from within a Makefile is ... not obvious to me at the moment :-(

robertinant, 2015-03-29:
Thanks @WestfW! I'll looked at this a while back and was not able to find a proper work-around. Will look into it again using the hints you posted here.

Thanks!

Robert

WestfW, 2015-03-29:
Potentially, if the IDE java code is already parsing pathnames, instead of using just strings, the conversion to the the 8.3 format could be done when writing Variables.mk.  http://stackoverflow.com/questions/11038595/java-call-for-windows-api-getshortpathname

KermMartian, 2015-08-13:
I've run into this issue as well with the MSP432 and Energia 0101e0016. Is there any potential fix in the pipeline?

robertinant, 2016-08-29:
The build process has completely changed in Energia 18 and fixes the issue with spaces in the path as described in this issue.

closed 2016-08-29 by robertinant