energia/Energia #588: Update the IDE to release 1.6.2 of Arduino

rei-vilo, 2015-03-28:
Release 1.6.2 of Arduino includes many nice features as:
* Library Manager
* Java 7 support
* Mac OS X native dialogs 
* ...

spirilis, 2015-03-28:
Easy 3rd party hardware support, which makes it much simpler to add new platforms... should speed along TI's own development efforts as they release new chips...
Great reasons to spend a few months sitting down and "starting fresh" with the Arduino 1.6.X IDE.

rei-vilo, 2015-03-30:
Release 1.6.2 of the Arduino IDE has also a mechanism to manage packages. A package contains all the material related to a platform. For example, the SAM package isn't installed by default.

The main benefit is a lighter initial download. Then the user installs the packages for the boards he/she uses.

[image]

On the downside, the standard ASCII files for boards, platforms and programmes are no longer under the folder `/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt` but they have moved to the hidden folder `/Users/ReiVilo/Library/Arduino15/packages/arduino/hardware/avr/1.6.2/boards.txt`.

rei-vilo, 2015-06-05:
Release 1.6.2 of the Arduino.CC inaugurates a new structure, incompatible with previous releases 1.0.6 and 1.5.8.

On Mac OS X, Arduino.CC 1.6.2 comes only with the framework for the 8-bit ATmega MCUs, with the standard location at `ArduinoCC.app/Contents/Java/hardware/arduino/avr`.

The other boards —32-bit Arduino Due, Intel Edison— are saved under the hidden folder `~/Library/Arduino15/packages/arduino`. This makes Arduino.CC 1.6.2 incompatible with previous releases. 

So I recommend keeping all the boards and their respective frameworks under one single folder, `Energia.app/Contents/Java/hardware`.

The **Boards Manager** is a good idea for a compact initial distribution of Energia with a limited set of boards. The end-user then selects the boards he/she wants to use.

rei-vilo, 2015-06-15:
The standard IDE based on Processing is being replaced by a more modern one, called [Arduino Studio](http://labs.arduino.org/tiki-index.php?page=Arduino%20Studio).

[image]

Among other features:

Features
>
Take advantage of Brackets editor's architecture;
* One editor for all environments: Cloud-based on your browser, Stand-alone on your PC and Embedded on your board;
* Written in Javascript and Node.js as Brackets extensions or as a stand-alone software;
* Brand new User Interface;
* Code Completion and in-line Quick Language Reference;
>

spirilis, 2015-06-15:
That's Arduino.ORG's replacement.  Are there indications Arduino.CC is doing the same?

rei-vilo, 2015-06-15:
This one comes from Arduino.ORG, but Arduino.CC is said to be working on something very similar, inspired by Cloud9 featured on the BeagleBone for the coming Arduino Tre.

rei-vilo, 2015-06-28:
Maybe **wait and see** is the better option.

rei-vilo, 2015-12-29:
See Readiness for Java SE 8 #533

rei-vilo, 2016-01-06:
Ticket updated with release 1.6.7 of Arduino.

ivazqueznet, 2016-07-08:
Regardless of what the Arduino _IDE_ does, it should still be possible to have Energia act as an Arduino core for MSP430 and Tiva C in the same way that cores for [STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) and [ESP8266](https://github.com/esp8266/arduino) exist for modern versions of the IDE. I understand that this means that Energia will become little more than a set of libraries, compilers, and configuration files, but shedding (or consolidating, in upstream) code responsibility is not necessarily a bad thing.

rei-vilo, 2016-07-08:
@ivazqueznet 

Pelase refer to [Energia 1.6.10E18B6 Beta available](http://forum.43oh.com/topic/9752-energia-1610e18b6-beta-available/).

robertinant, 2016-07-08:
@ivazqueznet, Yes, I will make the json file available that will allow you to install Energia cores such as msp430 and TivaC in the Arduino IDE. There are still a couple things I need to look at to resolve some issues with installing the cores into Arduino. When that is fixed, you can install Energia cores into the Arduino IDE just as you can with other none Arduino platforms.