energia/Energia #591: MSP432: "Compile DriverLib from source" preference causes failed compile.

WestfW, 2015-03-29:
The MSP432 support ni 0101E0015 includes a new preference item "Compile DriverLib from Source" item, but checking it results in a failed compile:

~~~~
/Applications/arduino/Energia-e15.app/Contents/Resources/Java/tools/common/bin/make --no-print-directory -C /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/build6157934834490470798.tmp/ -f /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/Makefile 
CCROOT /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f
SDKROOT /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/CC3200SDK_1.0.0.patch
CLOSURE /Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common
MAINSKETCH Blink.cpp
armcl Blink.cpp
/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/bin/arm-none-eabi-gcc -c -Os @"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/compiler.opt" -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/configPkg/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/src/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/avr/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/driverlib/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/inc/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/lib/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/configPkg/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/gnu/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/src/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/ti/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/xdc/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/AIR430BoostUSAFCC/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/AIR430BoostUSAFCC/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Adafruit_TMP006/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/CogLCD/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/EduBPMKII_Screen/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/M2XStreamClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/MQTTClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/OPT3001/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/PubNub/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/PubSubClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Temboo/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Temboo/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/WiFi/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/WiFi/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/aJson/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/aJson/utility/ -DF_CPU=48000000L -D__MSP432P401R__ -DTARGET_IS_MSP432P4XX -Dxdc__nolocalstring=1 -ffunction-sections -fdata-sections -DARDUINO=101 -DENERGIA=14 -D__MSP432P401R__ -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant  -fno-exceptions -fno-rtti -I "/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/include" -I "/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common" -I "/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common" -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/inc/CMSIS Blink.cpp -o Blink.obj
armcl main.cpp
/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/bin/arm-none-eabi-gcc -c -Os @"/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/compiler.opt" -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/configPkg/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/src/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/targets/MSP-EXP432P401R/ti/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/avr/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/driverlib/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/inc/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/cores/msp432/lib/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/configPkg/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/gnu/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/src/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/ti/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common/xdc/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/AIR430BoostUSAFCC/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/AIR430BoostUSAFCC/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Adafruit_TMP006/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/CogLCD/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/EduBPMKII_Screen/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/M2XStreamClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/MQTTClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/OPT3001/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/PubNub/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/PubSubClient/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Temboo/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/Temboo/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/WiFi/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/WiFi/utility/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/aJson/ -I/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/msp432/libraries/aJson/utility/ -DF_CPU=48000000L -D__MSP432P401R__ -DTARGET_IS_MSP432P4XX -Dxdc__nolocalstring=1 -ffunction-sections -fdata-sections -DARDUINO=101 -DENERGIA=14 -D__MSP432P401R__ -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant  -fno-exceptions -fno-rtti -I "/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/tools/lm4f/include" -I "/Applications/arduino/Energia-e15.app/Contents/Resources/Java/hardware/common" -I "/Application [... cut ...]
/db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/power/PowerMSP432_tirtos.c:62: undefined reference to `CPU_cpsid'
/db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/power/PowerMSP432_tirtos.c:110: undefined reference to `CPU_cpsie'
/db/vtree/library/trees/zumaapps/zumaapps-f28/products/drivers-next/packages/ti/drivers/power/PowerMSP432_tirtos.c:110: undefined reference to `CPU_cpsie'
collect2: error: ld returned 1 exit status
make: *** [Blink.cpp.elf] Error 1
/Applications/arduino/Energia-e15.app/Contents/Resources/Java/tools/common/bin/make returned 2
~~~~

robertinant, 2015-03-29:
Thanks for opening this issue! Will look into it this week!

d-c-d, 2015-10-30:
What's the status of this?
I just encountered an issue with building the libraries on the msp432 also.

closed 2016-08-29 by robertinant