energia/Energia #594: trying to begin with a static IP after DHCP fails locks up the program

jstamour802, 2015-03-31:
My program stops working if DHCP fails and I try to begin with a static IP. If you try to call ethernet.begin(mac,IP) after DHCP fails, the program locks up.

The "DHCP chat server example" demonstrates this - if you have no DHCP it never begins with the static IP and moves beyond the setup loop

void setup() {
   // open the serial port
  Serial.begin(9600);
  Serial.println("\n\nDhcpChatServer setup");
  // start the Ethernet connection:
  Serial.println("Trying to get an IP address using DHCP");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");  
    // initialize the ethernet device not using DHCP***************never gets past this point!!!*************
    Ethernet.begin(mac, ip, gateway, subnet);
  }
  // print your local IP address:
  Serial.print("My IP address: ");
  ip = Ethernet.localIP();
  for (byte thisByte = 0; thisByte < 4; thisByte++) {
    // print the value of each byte of the IP address:
    Serial.print(ip[thisByte], DEC);
    Serial.print("."); 
  }
  Serial.println();
  // start listening for clients
  server.begin();
 
}

rei-vilo, 2015-03-31:
Which MCU?

a-bhat1, 2015-03-31:
MCU = MSP430F5529
Board = MSP430F5529 Launch Pad

Regards,
Arthi

rei-vilo, 2015-03-31:
Thanks. So I guess you're using the CC3100 BoosterPack.

a-bhat1, 2015-03-31:
Well, the USBHID is a library that provides a HID interface to the target board.  You can use an applicable booster pack that is for the MSP430F5529 launch pad and create your sketch/application and then use the USBHID interface to send data between the MSP430F5529 launch pad and the host computer using USB.

Currently, the example that is included in the USBHID library does not require a booster pack.  The sketch is designed to look for data at the HID interface and then send and receive data to/from the MSP430F5529 launch pad.

In order to send and receive the HID data on the host side, you will need to download the HidDemo tool that is packaged with the MSP430 USB Developers Package.

Regards,
Arthi

jstamour802, 2015-03-31:
I'm usingthe TM4C1294XL connected launchpad and have this problem

robertinant, 2016-08-29:
This issue was moved to energia/tivac-core#10

closed 2016-08-29 by robertinant