energia/Energia #598: emt: empty folder or dead link

rei-vilo, 2015-04-07:
The `emt @ 602e5f9` folder on the `master` branch is empty or the `emt @ 602e5f9` link is dead.

rei-vilo, 2015-04-07:
I compiled **Energia** from the downloaded GitHub repository with the standard `ant dist`.

The resulting IDE fails to compile, as many core libraries —including `Arduino.h`— are missing.
[image]

Comparing the home built Energia with the official distribution shows many files are just gone. Are they supposed to be in the `emt` missing folder or dead link?
[image]

rei-vilo, 2015-04-08:
From http://forum.43oh.com/topic/8347-energia-15-built-from-source-wont-compile-msp432/?p=65059
>
The Energia repository has a temporary submodule called emt that points to a private repository on a different server. This repository will eventually be merged into the Energia main repository. The files that are "missing" are symbolic links to that repository. Hence if the content of the submodule is missing then the files are missing in the Energia repo. To work around this for now, you could copy the files from the release to the Energia repository. The files to copy are hardware/common/* and hardware/msp432/*
>
I hope to have these files in the main Energia repository later this month.
> 
Hope that helps.
Robert
>

rei-vilo, 2015-04-17:
Still dead link or missing folder after commit 	69b57433b73820b0aadd2bb9dd4415f7c2cf18ee.
[image]

robertinant, 2016-08-29:
The emt submodule is no longer used in Energia 18. The core is not located in it's separate package and repository.

closed 2016-08-29 by robertinant