energia/Energia #600: Hi everyone I have some problems with cc3200 wifi and AD0(GPIO02).My energia version is E0015.

JimStein, 2015-04-09:
I try to run a code which can transmit sensor values through WIFI. I use Matlab to collect the data from cc3200 through router. It works fine with AD1/AD2/AD3. But it always returns 1FF with AD0 no matter what signals I connected to GPIO02. I noted that AD0 has a multi-use as serial RX is there any affections? I do not need serial functions is there any method to avoid this problem? I do want to use AD0. Is anyone have some ideas? Thank you very much!

JimStein, 2015-04-09:
The following is my codes. Hope anyone can solve my problems . I'm very appreciate.

/*
  WiFi UDP Send and Receive String

 This sketch wait an UDP packet on localPort using the CC3200 launchpad
 When a packet is received an Acknowledge packet is sent to the client on port remotePort

 created 30 December 2012
 by dlf (Metodo2 srl)
 
 modified 1 July 2014
 by Noah Luskey

 */

#ifndef __CC3200R1M1RGC__
// Do not include SPI for CC3200 LaunchPad
#include <SPI.h>
#endif
#include <WiFi.h>

// your network name also called SSID
char ssid[] = "A438";
// your network password
char password[] = "sensor123";

unsigned int localPort = 8000;      // local port to listen on
//int sensorPin = A0;    // select the input pin for the potentiometer
uint16_t sensorValue = 0;  // variable to store the value coming from the sensor
char packetBuffer[255]; //buffer to hold incoming packe
char  ReplyBuffer1[] = "00";       // a string to send back
char ReplyBuffer2[] = "00";       // a string to send back
char ReplyBuffer3[] = "00";       // a string to send back
char ReplyBuffer4[] = "00";       // a string to send back

const int NTP_PACKET_SIZE = 6;

WiFiUDP Udp;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);

  // attempt to connect to Wifi network:
  Serial.print("Attempting to connect to Network named: ");
  // print the network name (SSID);
  Serial.println(ssid); 
  // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
  WiFi.begin(ssid, password);
  while ( WiFi.status() != WL_CONNECTED) {
    // print dots while we wait to connect
    //Serial.print(".");
    delay(300);
  }
  
  Serial.println("\nYou're connected to the network");
  Serial.println("Waiting for an ip address");
  
  while (WiFi.localIP() == INADDR_NONE) {
    // print dots while we wait for an ip addresss
  //  Serial.print(".");
    delay(300);
  }

  Serial.println("\nIP Address obtained");
  //printWifiStatus();

 // Serial.println("\nWaiting for a connection from a client...");
  Udp.begin(localPort);
}

void loop() {

  // if there's data available, read a packet
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
  //  Serial.print("Received packet of size ");
  //  Serial.println(packetSize);
  //  Serial.print("From ");
    IPAddress remoteIp = Udp.remoteIP();
  //  Serial.print(remoteIp);
  //  Serial.print(", port ");
  //  Serial.println(Udp.remotePort());

    // read the packet into packetBufffer
    int len = Udp.read(packetBuffer, 255);
    if (len > 0) packetBuffer[len] = 0;
  //  Serial.println("Contents:");
  //  Serial.println(packetBuffer);

    // send a reply, to the IP address and port that sent us the packet we received
    Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
    
    
    sensorValue = 0;
    sensorValue = analogRead(A0);
    ReplyBuffer2[0] = sensorValue>>8;
    ReplyBuffer2[0] = ReplyBuffer2[0]|0x10;
    ReplyBuffer2[1] = sensorValue;
    if(sensorValue==0){
    Udp.write(ReplyBuffer2[0]);
    Udp.write(ReplyBuffer2[1]);
    }
    else{
    Udp.write(ReplyBuffer2);
    }
    
    Udp.endPacket();
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

JimStein, 2015-04-09:
Also share you with my matlab code .

port = 8000;
ip=['192.168.5.143'];
   obj1 = instrfind('Type', 'udp', 'RemoteHost', ip, 'RemotePort', port, 'Tag', '');
if isempty(obj1)
   obj1 = udp(ip, port);
else
    fclose(obj1);
    obj1 = obj1(1);
end
set(obj1,'Timeout',0.3);
fopen(obj1);
fprintf(obj1, 're');
datatest = fread(obj1, 2,'uint16');
tic
sensor1=[];sensor2=[];sensor3=[];
for i=1:1:2000

fprintf(obj1, 're');
datatest = fread(obj1, 2,'uint16');
try
bitshift(datatest(1,:),-12);  %%通过右移判断是哪个传感器的读数
sensor=bitand(datatest(1,:),4095);   %%去除标志位
sensor1=[sensor1;sensor];
try
plot(sensor1(end-100:end,:),'.b');
drawnow;
catch
plot(sensor1,'.b');
drawnow;
end
hold on;
bitshift(datatest(2,:),-12);  %%通过右移判断是哪个传感器的读数
sensor=bitand(datatest(2,:),4095);   %%去除标志位
sensor2=[sensor2;sensor];

try
plot(sensor2(end-100:end,:),'.r');
drawnow;
catch
plot(sensor2,'.r');
drawnow;
end
bitshift(datatest(3,:),-12);  %%通过右移判断是哪个传感器的读数
sensor=bitand(datatest(3,:),4095);   %%去除标志位
sensor3=[sensor3;sensor];
try
plot(sensor3(end-100:end),'.g');
drawnow;
catch
plot(sensor3,'.g');
drawnow;    
end
catch
fprintf(obj1, 're');
end

end
toc

spirilis, 2015-04-09:
Be aware, when used in ADC mode, the CC3200 can only tolerate voltages up to 1.47V or so.  Above that it can be damaged.  Not sure if that's happened to you yet - what kind of voltage levels have you sent the ADC pin?

JimStein, 2015-04-09:
Thank you for your information . I had connected the GPIO2 with Agilent Power Sources and I'm quite sure my voltage is below the upper limit . Actually my sample voltage is ranging from 0 to 1V . But even the voltage is 0V AD0 still gives the answer of 1FF . It quite confused me. At the same time the other ADC is working rightly. Is there any possibility that AD0 crashed with serial or wifi?

spirilis, 2015-04-09:
I'm guessing it's burned out, yes.  I think that's the typical behavior when it's trashed.  @rei-vilo you've had this happen before, does that sound right?

rei-vilo, 2015-04-09:
Yes, it seems so. I fried one CC3200 when playing with the Kentec 3.5" SPI screen. I call those fried CC3200s **zombies**.

In my case, pin 23 was used.
>
A basic `analogRead(23)` always gives 4095, even if the pin is floating or connected to ground. Other analog inputs on pins 2, 6 and 24 return correct values.
>

spirilis, 2015-04-09:
although, 0x1FF isn't 4095.

rei-vilo, 2015-04-09:
> although, 0x1FF isn't 4095.

Circuitry might be fried with a specific value.

JimStein, 2015-04-10:
It's really great someone can discuss the question with me. Sorry for a delay reply . May be we have 12 hours time difference I guess :) . I followed your advice and I tested the AD0 again .  However I didn't use Energia this time .Instead I using CCSv6.1 to debug AD0  . Amazingly AD0 works quite right and return the data from serial when I connected GPIO2 to my power source(I use only Tx to send messages).

JimStein, 2015-04-10:
[image]

JimStein, 2015-04-10:
//*****************************************************************************
//
// Copyright (C) 2014 Texas Instruments Incorporated - http://www.ti.com/ 
// 
// 
//  Redistribution and use in source and binary forms, with or without 
//  modification, are permitted provided that the following conditions 
//  are met:
//
//    Redistributions of source code must retain the above copyright 
//    notice, this list of conditions and the following disclaimer.
//
//    Redistributions in binary form must reproduce the above copyright
//    notice, this list of conditions and the following disclaimer in the 
//    documentation and/or other materials provided with the   
//    distribution.
//
//    Neither the name of Texas Instruments Incorporated nor the names of
//    its contributors may be used to endorse or promote products derived
//    from this software without specific prior written permission.
//
//  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS 
//  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT 
//  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
//  A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT 
//  OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
//  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT 
//  LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
//  DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
//  THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT 
//  (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE 
//  OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
//
//*****************************************************************************

//*****************************************************************************
//
// Application Name     - ADC
// Application Overview - The application is a reference to usage of ADC DriverLib 
//                        functions on CC3200. Developers/Users can refer to this 
//                        simple application and re-use the functions in 
//                        their applications.
//
// Application Details  -
// http://processors.wiki.ti.com/index.php/CC32xx_ADC
// or
// docs\examples\CC32xx_ADC.pdf
//
//*****************************************************************************

//*****************************************************************************
//
//! \addtogroup adc_example
//! @{
//
//*****************************************************************************

// Standard includes
#include <string.h>
#include <stdint.h>
#include <stdlib.h>
#include <stdbool.h>

// Driverlib includes
#include "utils.h"
#include "hw_memmap.h"
#include "hw_common_reg.h"
#include "hw_types.h"
#include "hw_adc.h"
#include "hw_ints.h"
[... 194 lines cut ...]

        uiIndex = 0;

        UART_PRINT("\n\rTotal no of 32 bit ADC data printed :4096 \n\r");
        UART_PRINT("\n\rADC data format:\n\r");
        UART_PRINT("\n\rbits[13:2] : ADC sample\n\r");
        UART_PRINT("\n\rbits[31:14]: Time stamp of ADC sample \n\r");

        //
        // Print out 4096 ADC samples
        //
        while(uiIndex < 4096)
        {
            UART_PRINT("\n\r0x%x\n\r",pulAdcSamples[uiIndex++]);
        }

    }

}
//*****************************************************************************
//
// Close the Doxygen group.
//! @}
//
//*****************************************************************************

JimStein, 2015-04-20:
Thank you for everyone. I have finished my debug.Though I didn't solve this question with Energia E0015. However, I use CCS to code it and finally all 4 ADCs work functionally through WIFI.  Sooner or later I will check the source code in Energia and try to raise the question. I will return my discover in this page.  At last , thank you again. Have a nice day!

pavelh2000, 2015-06-03:
Hi Jim,
I understand that ch0 can be functioned using other platforms rather than energia? Did you check the source code in energia? Because i have the same problem as i am trying to get 4 adc from cc3200 using energia. Your feedback will be appreciated. Thanks.

rei-vilo, 2016-03-13:
How does the problem behave with more recent release of Energia, e.g. 17?