energia/Energia #601: MSP430F5529.h TA2IV Definitions

SvdSinner, 2015-04-13:
The MSP430F5529.h beginning at line 3218 reads:
```
/* TA2IV Definitions */
#define TA2IV_NONE          (0x0000)    /* No Interrupt pending */
#define TA2IV_TA1CCR1       (0x0002)    /* TA2CCR1_CCIFG */ 
#define TA2IV_TA1CCR2       (0x0004)    /* TA2CCR2_CCIFG */ 
#define TA2IV_3             (0x0006)    /* Reserved */
#define TA2IV_4             (0x0008)    /* Reserved */
#define TA2IV_5             (0x000A)    /* Reserved */
#define TA2IV_6             (0x000C)    /* Reserved */
#define TA2IV_TA2IFG        (0x000E)    /* TA2IFG */
```

But, based on the A0, and A1 timer interrupt definitions, it should read:
```
/* TA2IV Definitions */
#define TA2IV_NONE          (0x0000)    /* No Interrupt pending */
#define TA2IV_TA2CCR1       (0x0002)    /* TA2CCR1_CCIFG */
#define TA2IV_TA2CCR2       (0x0004)    /* TA2CCR2_CCIFG */ 
#define TA2IV_3             (0x0006)    /* Reserved */
#define TA2IV_4             (0x0008)    /* Reserved */
#define TA2IV_5             (0x000A)    /* Reserved */
#define TA2IV_6             (0x000C)    /* Reserved */
#define TA2IV_TA2IFG        (0x000E)    /* TA2IFG */
```
(Issue is in the second part of the interrupt definitions for CCR1 and CCR2)

spirilis, 2015-04-14:
That probably goes back to the original source of the toolchain used, mspgcc 4.6.3 (no longer maintained).

Fwiw, the new RedHat GCC includes headers like this-
```
/* TA2IV Definitions */
#define TA2IV_NONE          (0x0000)    /* No Interrupt pending */
#define TA2IV_TACCR1        (0x0002)    /* TA2CCR1_CCIFG */
#define TA2IV_TACCR2        (0x0004)    /* TA2CCR2_CCIFG */
#define TA2IV_3             (0x0006)    /* Reserved */
#define TA2IV_4             (0x0008)    /* Reserved */
#define TA2IV_5             (0x000A)    /* Reserved */
#define TA2IV_6             (0x000C)    /* Reserved */
#define TA2IV_TAIFG         (0x000E)    /* TA2IFG */

/* Legacy Defines */
#define TA2IV_TA2CCR1      (0x0002)    /* TA2CCR1_CCIFG */
#define TA2IV_TA2CCR2      (0x0004)    /* TA2CCR2_CCIFG */
#define TA2IV_TA2IFG       (0x000E)    /* TA2IFG */
```

SvdSinner, 2015-04-14:
I suspect the overlap of people who notice things, people who rebort bugs, and people who use Timer A2 (Rather than A0 or A1) isn't very big.  :)

spirilis, 2015-04-14:
Yeah, that's probably true ;)  And for the most part GCC doesn't take advantage of the potential performance gain from the peripheral Interrupt Vector registers like how TI's cl430 compiler does with its switch(__even_in_range()) construct.

robertinant, 2016-07-18:
This has been corrected a while back when I upgraded the headers.

closed 2016-07-18 by robertinant