energia/Energia #602: usbserial frequency

gatoMSP430, 2015-04-15:
I am trying to use usbserial library with a custom made msp430f5529 board. XT2 has 24MHz frequency. I know for sure it can establish USB communication, because C1 example from usb developers package compiles and works, as long as I change USB_XT_FREQ and USB_XT_FREQ_VALUE in line 102,103 of descriptors.h of the ccs project. The same change, in descriptors.h of the energia usbserial library makes no effect. 
I have checked VUSB pin's voltage manually, and while the ccs project turns that on (3.3V), the energia project turns it on, and then decides to turn it off (as if it had decided the PLL is externally feeded, which it is not). 
Any ideas?

rei-vilo, 2015-04-15:
Maybe the 43oh.com forum is a better place to get your question answered.

gatoMSP430, 2015-04-15:
It is an issue related to the energia implementation of usbserial, so I think it might be the right place

a-bhat1, 2015-04-15:
Hi,
Maybe changing the value being passed in for XT2_Start() in USBSerial.cpp file from XT2DRIVE_0 to XT2DRIVE_3 might help.   If you look in usb.c file in USB_enable() function,  for 24MHZ, the XT2Drive should be at level 3.

Regards,
Arthi Bhat

gatoMSP430, 2015-04-15:
Thanks for the answer. I had tried that. Actually, I have tried to eliminate any options not involving msp430f5529, one by one. But it doesn't work. I wonder wetter it is related to the different architecture. the launchpad uses a usb hub, connecting to either the flash tool or the f5529's usb

StefanSch, 2015-04-16:
There are 3 lines in the descriptors.h file which are controlling the crystal setting for the USB PLL

Please check if this are set to the right values:
#define USB_PLL_XT        2                  // Defines which XT is used by the PLL (1=XT1, 2=XT2)
#define USB_XT_FREQ_VALUE       4.0   // Indicates the freq of the crystal on the oscillator indicated by USB_PLL_XT
#define USB_XT_FREQ       USBPLL_SETCLK_4_0  // Indicates the freq of the crystal on the oscillator indicated by USB_PLL_XT

For a 24MHz crystal at XT2 it should be (without testing):
#define USB_PLL_XT        2                  // Defines which XT is used by the PLL (1=XT1, 2=XT2)
#define USB_XT_FREQ_VALUE       24.0   // Indicates the freq of the crystal on the oscillator indicated by USB_PLL_XT
#define USB_XT_FREQ       USBPLL_SETCLK_24_0  // Indicates the freq of the crystal on the oscillator indicated by USB_PLL_XT

gatoMSP430, 2015-04-16:
I tried those, too. I am debugging the energia script with ccs, to check the registers. One of the things I noticed is execution step by step was getting trapped at the loop at HAL_UCS, at line 209, checking SFRIFG1&OFIFG. After commenting that, it does execute. The device is detected at some point, but not recognized. Checked USBPLLDIVB register, and it had the right value for 24MHz (0x050F). XT2 is enabled, and ok. I did not soldet XT1, as I didn't need one. I'll keep checking

spirilis, 2015-04-16:
SFRIFG1 & OFIFG may be set for any oscillator faults.  If XT1 isn't soldered, be sure XT1OFF is set in UCSCTL6, then leave that statement uncommented in HAL_UCS.

spirilis, 2015-04-16:
Keep in mind, Energia clears XT1OFF and never re-sets it if it finds a timeout when trying to spin that up.  Energia prefers to use XT1 to drive the FLL for the DCO but if it can't get XT1 to run, it will revert to the built-in REFO.

Or try soldering XT1; your startup times will be faster since Energia tries to start XT1 whether you like it or not, only now it won't have to go through the ~1 second timeout period.

gatoMSP430, 2015-04-17:
I soldered xt1 as suggested.that solved part f.the problem.checking registers showed xtrdrive_0 was imposed in several places for xt2.after checking them all,it worked.thanks for the help

spirilis, 2015-04-18:
Sweet, enjoy!

closed 2015-04-18 by spirilis