energia/Energia #603: Serial.println behavior change

treygerman, 2015-04-15:
Starting in version 15 of Energia, the behavior of the Serial.println function has changed.  Previously calling this function with a variable would convert the value of the variable to ASCII before sending it out the serial port.  In version 15 the variable's value is interpreted as a single ASCII character.

Version 14:
int unicornPoop = 42;
Serial.println(unicornPoop); // Prints "42" 

Version 15:
int narwhalFarts = 69;
Serial.println(narwhalFarts); //Prints "E"

Given the documentation on Energia.nu, this function needs to be changed back to operating the way it did previously.

spirilis, 2015-04-15:
Which platform does this?

treygerman, 2015-04-15:
Oh yea...guess that's important.  I've seen this on MSP430 and MSP432.

spirilis, 2015-04-15:
Energia 15 on Linux, FR5969 V2.0 LaunchPad -
```
void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop()
{
  // put your main code here, to run repeatedly:
  static int unicornShit = 48;  // ASCII 48 is '0'
  
  Serial.println(unicornShit++);
  delay(500);
}
```

Serial monitor output:
```
48
49
50
51
52
53
54
55
56
57
58
59
60
48
49
50
51
52
53
54
55
56
57
58
59
60
61
62
```

treygerman, 2015-04-15:
Try changing unicornShit to a char...

spirilis, 2015-04-15:
That would do it, but that's expected behavior!  The print/println functions are overloaded by data type:
(from hardware/msp430/cores/msp430/Print.h)
```
    //size_t print(const __FlashStringHelper *);
    size_t print(const String &);
    size_t print(const char[]);
    size_t print(char);
    size_t print(unsigned char, int = DEC);
    size_t print(int, int = DEC);
    size_t print(unsigned int, int = DEC);
    size_t print(long, int = DEC);
    size_t print(unsigned long, int = DEC);
    size_t print(double, int = 2);
    size_t print(const Printable&);

    //size_t println(const __FlashStringHelper *);
    size_t println(const String &s);
    size_t println(const char[]);
    size_t println(char);
    size_t println(unsigned char, int = DEC);
    size_t println(int, int = DEC);
    size_t println(unsigned int, int = DEC);
    size_t println(long, int = DEC);
    size_t println(unsigned long, int = DEC);
    size_t println(double, int = 2);
    size_t println(const Printable&);
    size_t println(void);
```

println(char) from Print.cpp:
```
size_t Print::println(char c)
{
  size_t n = print(c);
  n += println();
  return n;
}
```
(since 'c' in that function is already defined in the function arguments as char, the compiler knows to call the 'char' variety of print())

print(char) from Print.cpp:
```
size_t Print::print(char c)
{
  return write(c);
}
```
The only single-argument, single-byte variant is write(uint8_t), so I'm guessing the C++ compiler chooses that as the closest possible "variant", and it treats the value literally in whatever context the upper layer library deems appropriate (HardwareSerial extends Stream which extends Print, so write(uint8_t) goes to HardwareSerial::write(uint8_t), which adds the byte value to the TX buffer where it's sent out the UART)

spirilis, 2015-04-15:
Ok so there is an oddity here that's worth mentioning.

char on msp430 and possibly ARM platforms is, by default, "signed char", i.e. int8_t
print/println(char) then handles that as if it were an ASCII character.

However, uint8_t or "unsigned char" is treated as a number and spelled out:
(Print.h)
```
    size_t print(unsigned char, int = DEC);
```
(Print.cpp)
```
size_t Print::print(unsigned char b, int base)
{
  return print((unsigned long) b, base);
}
```
(Print.cpp)
```
size_t Print::print(unsigned long n, int base)
{
  if (base == 0) return write(n);
  else return printNumber(n, base);
}
```
printNumber of course expands it out into a sequence of numeric ASCII characters as necessary to represent the number in text...

So that's an interesting thing, that char vs. unsigned char is treated differently.

spirilis, 2015-04-15:
This behavior is identical on the lm4f and cc3200 (same code used for Print.cpp/Print.h) and presuming char = signed int8 and unsigned char = unsigned int8 on C2000, should be the same way on the C2000 port.

treygerman, 2015-04-15:
Interesting...ya learn something new everyday

closed 2015-04-15 by treygerman