energia/Energia #604: aJson library // char* payload = aJson.print(msg) not working

adrianF-TI, 2015-04-17:
I have replicated this on both 432 & F5529LP LaunchPad using Energia v15. The aJson.print(msg) function will work for a bit, but will eventually start printing out rubbish. 

Note however that aJson.print(msg, &stream); works just fine & continues to work even when the aJson.print(msg) fails to.

I want aJson.print(msg) to work because I want to save the string into a variable so that I can publish it via MQTT. The aJson.print(msg, &stream) only sends the string to the terminal.

adrianF-TI, 2015-04-21:
User error! aJson.print(msg) uses malloc & returns a pointer. Need to call free(msg) in the main loop of your sketch to free up the pointer to the character array once you're done using it.

closed 2015-04-21 by adrianF-TI

spirilis, 2015-04-21:
ugh, I hate having library functions that malloc() without taking full ownership of the end result. (</rant>)