energia/Energia #607: Provide Guidelines to Add Support for New MCU

rei-vilo, 2015-04-25:
There are numerous tickets related to `Add Support for New MCU`.

As Energia –and its team– can't obviously add support for all the MCUs and variants from the extensive Texas Instruments portfolio, why not provide guidelines to help enthusiasts to do so?

More specifically:
* Structure of the folders 
* List of files impacted
* Conventions

rei-vilo, 2015-04-25:
All mentioned tickets #596 #541 #350 #214 #169 could be closed.

astuder, 2015-05-01:
There are also a few related pull-requests: #309 #344 #421 
I haven't touched mine (#344) in ages, so I won't mind if you just close it as well.

rei-vilo, 2015-05-01:
@astuder 
I let the decision to Robert.

rei-vilo, 2016-07-13:
Energia is derived from Arduino, so the all  the Arduino resources apply for Energia.

Please find the reference at the [Arduino wiki](https://github.com/arduino/Arduino/wiki/_pages).

Another useful place is the [Energia section on the 43oh forum](http://forum.43oh.com/forum/119-energia/).

rei-vilo, 2016-08-16:
See #912