energia/Energia #611: Delays in the Wire library

elluna, 2015-05-07:
I'm using Energia 0101E0014 with a Tiva-C Launchpad and using the Wire library to connect to some peripherals.
My code worked initially with Arduino Mega and then I ported it to the Launchpad. 
I realized that the access to the peripherals was much slower than with Arduino. Having a look to the Wire.cpp source code I found that the functions getRxData() and sendTxData() include a delay(1), that means that every time they're called, a millisecond delay is introduced.
I simply commented out those delays and now works fine.
What are these delays for?
Should I create a new bug fix to remove such delays?

Regards,
Eduardo

rei-vilo, 2015-05-07:
I guess those delays are required. 

Have you tried I²C in fast mode (400 kHz) and fast mode+ (1 MHz)?

elluna, 2015-05-08:
Hello,
Required for what?
I've just removed those delays and it is working perfectly with 4 devices connected.
The speed problem is not related with the transmission speed, it is because the Tx and Rx functions waits for 1 ms before doing its job and I don't understand what this delay is for, because in all my testing is not required.

robertinant, 2015-09-24:
The delays are indeed there get the correct behavior. With that said, it is not the right way to do it. Instead of delay, it should check if the master is busy. I have replaced the delays with while(ROM_I2CMasterBusy(MASTER_BASE));

closed 2015-09-24 by robertinant