energia/Energia #618: Cannot launch Java application error message

lateAtNight, 2015-05-10:
When opening Energia 15 0101E0015 for OSX I get this error message:

"Cannot launch Java application UnCaught exception in main method: java.lang.ArithmeticException: / by zero". 

The only option on the dialogue box is to quit. Operating system is OSX 10.10.4 on a MacBook Pro. Only a few days ago it was working...
[image]

rei-vilo, 2015-05-10:
I guess it was working with Mac OS X 10.10.3!

honteng, 2015-05-25:
My OS X is 10.10.3 and I see the same error.

honteng, 2015-05-25:
rm -rf ~/Library/Energia does work for work around

lateAtNight, 2015-05-25:
honteg: your solution works on 10.14, thank you very much! I was considering going to the extreme measure of reformatting...

rei-vilo, 2015-06-28:
Upgrading to Java 8 may solve this issue. See #533

davidbuzz, 2015-09-11:
I just had this issue on Energia 16 on OSX

I found that by deleting or renaming my ~/Library/Energia/preferences.txt, then the problem went away.    

I also tried ( as an alternative) editing my preferences.txt and I deleted all the lines that started with "last", and that also fixed the issue. ( do this with the Energia app closed).

I can confirm that both of these fixed the problem, and that by restoring the original preferences.txt file, then the problem re-manifests, so it's not just speculation.