energia/Energia #621: Add remoteIP() Function to WiFi Library

rei-vilo, 2015-05-18:
The `remoteIP()`function is available for [Ethernet UDP](http://energia.nu/reference/ethernet/ethernet_udpremoteip/) and [Wifi UDP](http://energia.nu/reference/wifi/wifi_wifiudpremoteip/) but not for WiFi.

rei-vilo, 2015-05-19:
This also concerns the CC3100 BoosterPack.

robertinant, 2015-05-19:
This library was implemented according to the original Arduino WiFi library which does not implement remoteIP() for TCP. What would you use the remote IP address for?

spirilis, 2015-05-19:
I hadn't looked at that much but I am surprised even Arduino's lib doesn't include it.  Being able to verify the connected peer would be nice for security purposes, or logging.

rei-vilo, 2015-05-19:
I've used the `remoteIP()` on the ESP8266 and wanted to use it also with the CC3200.

robertinant, 2015-05-19:
Can you point me to the sourced for the ESP WiFi driver?

spirilis, 2015-05-19:
Should be here - https://github.com/esp8266/Arduino/tree/esp8266/libraries/WiFi/src - but I don't see any remoteIP() in WiFiClient.cpp ... just WiFiUdp.cpp

rei-vilo, 2015-05-19:
Sure, please have a look at [WiFiClient.h](https://github.com/esp8266/Arduino/blob/202d38bb3e98d450710abb11823b0db3dab57c66/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h) and [WiFiClient.cpp](https://github.com/esp8266/Arduino/blob/202d38bb3e98d450710abb11823b0db3dab57c66/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp) from the [ESP8266 / Arduino](https://github.com/esp8266/Arduino) project.

`remoteIP()` is also listed on other [documents](https://github.com/esp8266/Arduino/search?p=1&q=remoteip) on the same repository.

rei-vilo, 2015-05-19:
@spirilis You need to look at `Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp`, not `Arduino/libraries/WiFi/src/WiFiClient.cpp`.

spirilis, 2015-05-19:
Anyway regardless, my 2 cents is that it's a valuable function to have, phooey on Arduino for messing up by not including it in their version ;-)