energia/Energia #622: Serial ports not listed on Energia 0015 Mac OS

flety-emmanuel, 2015-05-19:
Having issues with OSX now we're deploying our boards within the team with collaborators equipped with macbook computers.

Our CC3200 (mod) design doesn't use a FTDI chip but a silicon labs CP2102. We've installed the driver from silabs, it's signed. Other apps see the port without problem, they appear in /dev with tty. and cu. Obviously it's working just fine under windows.

We've tested this on OSx yosemite and also 10.9 : arduino (1.6) sees the uart port, max msp as well, but the serial port menu remain greyed in energia 0015. The only moment we can see something in the serial port menu is when I plug an TI CC3200 launchpad (with the proper driver installed).

Normally, we should see there everything related to a com port, which includes the bluetooth port, or any USB uart, just like in arduino but that's not the case. So we can't upload code.

That seems to be the case in the wiki here, in the serial port selection paragraph
https://github.com/energia/Energia/wiki/Getting-Started

We installed various older releases of Energia, it works in version 009, all ports are listed, but not in version 0011 and further. Urgent, we're leaving for our hack even in a few days.
Please Help. I don't have the tool chain to modify this myself in the code nor to recompile

rei-vilo, 2015-05-19:
Have you tried uploading to the CC3200 with CP2102 using the command line on a terminal window?

flety-emmanuel, 2015-05-19:
thanks. No I haven't tried this, very good, I'll give it a go. We have selected energia to be able to merge with the "duino" community. It's mean to be a DIY / dev platform, not a product. I just don't understand why all sorts of serial ports aren't listed anymore.

flety-emmanuel, 2015-05-19:
The command line works with the launch pad. When I force the command line using the other com port cu.SLABS.... it says can't open the device, then "uart open" "open success" then ACK fails. Overall, it refuses to open the port, no idea why. It's like it's been hard coded to certain port name or VID or PID

rei-vilo, 2015-05-19:
You can try `openocd` to upload the binary, provided the driver for the CP2102 is available.

robertinant, 2015-05-19:
The Energia IDE filters for the following Serial port names:

Editor.java
```
1094           if (Base.isMacOS()) {
1095                 if(curr_port.indexOf("tty.uart") == -1
1096                 && curr_port.indexOf("tty.usbmodem") == -1
1097                 && curr_port.indexOf("tty.usbserial") == -1
1098                 && curr_port.indexOf("cu.usbserial") == -1) continue;
1099           }
```
Since the silibs port is named cu.SLABS it is filtered out and not displayed. I could add the cu.SLABS to the list of accepted ports in the next release but given that you can not upload using the command line either we need to look into that before we do anything else.

Can you copy past the entire output of the cc3200prog when attempting to upload?

flety-emmanuel, 2015-05-19:
very cool help from both of you rei-vilo and robertinant, thank you. I'm back home with no access to the mac right now but it goes like if the bootloading process fails when cc3200 doesn't receive the ACK from the board, same messages PLUS "can't open device" before everything. It seems that it tries to open the port device, fails (but not like the port was already opened, that leads to the usual error message), then it say "uart open" and "open success" or something similar (from memory).

It's really like cc3200prog is also filtering something. Glad to see that at least there's something "logical" in the editor that filters only certain names. That would be ideal to add the silicon labs tty too but I think any serial port should be detected. In our case, even the bluetooth port (which I think shows up as bt-modem something) isn't listed while it was in the past. I think I've tried with a usb-serial cable from FTDI and I don't think I could see it neither, I'll test tomorrow

robertinant, 2015-05-19:
I just looked at the cc3200prog sources and can't spot anything obvious. The only thing I can thing of is that the ACK timeout is to short. We have seen this when trying to upload large binaries. What is the size of the binary that you are trying to upload? Seeing the output of cc3200prog would be really helpful.

flety-emmanuel, 2015-05-19:
I did this test with the blink tutorial / ex. sketch which is like a few kb. I noticed (on windows) the ACK issue sometimes when trying to upload my current work sketch which is 100 kb (ish) which requires to manually keep the module in reset and unleash it when energia moves from compile to upload. A longer timeout for the ACK would be apreciated.

I'll send the log tomorrow morning first thing, the mac is at the lab. Thanks again for your hard work and prompt answers

On a side note, I suppose you're still hands tied with the cc3200prog source code. Our platform would definitely have some good use of an uploader that could deal with a simple UART vs the complex things achieved by the launchpad double uart or the mk20 interface of redbearlabs. I don't know what's possible but having for instance the uploader program controlling a flux I/O of the uart like CTS or similar to control SOP2 would be much simpler than relying on a complex interface chip. In our case, we're using the arduino auto reset technique with a capacitor on DTR, which works greate most of the time (when we don't have the ack timeout issue), our only issue is the control of SOP2 which remains manual and which is PITA.

I don't know what is going to be the best practice for this. I foresee a lot of variants and board designs using the CC3200 in a very near future and I can't imagine not having a more versatile uploader (no offense meant) or at least a possibility to upload to a device with just a simple UART and not all the bells & whistles.

Thanks again

flety-emmanuel, 2015-05-20:
here is the result of the upload test. I can see that the port is indeed not opened as my device doesn't auto reboot. Also, that works just fine within arduino, on the same machine (10.9). The serial port monitor opens the port (my device auto reboots) and print info is displayed. The arduino serial port menu lists all my /dev/cu.xxxxxxx ports 
crw-rw-rw-  1 root  wheel   18,   1 19 mai 16:05 cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   3 19 mai 16:05 cu.Bluetooth-Modem
crw-rw-rw-  1 root  wheel   18,  17 19 mai 16:28 cu.SLAB_USBtoUART

all those should be seen in energia. The filter I sort of understand to avoid the serial port menu to be populated with everything you have on the machine, but on the CC3200 side, you just have a simple UART. In between can be *anything* including a wireless transmission, bluetooth, a custom device, and why not an arduino relaying the data. Listing everything tty and cu (or maybe just cu like arduino now does) sounds logical to me, like it was back in version 009

I don't know why cc3200prog proceeds after failing to open the port. Without the port opened, it's obvious that it will get a ACK timeout (which again should be longer to allow slower boot / reset time of the CC3200)
Thanks

cc3200prog log:

m1959:Energia flety$ /Applications/Energia-0015.app/Contents/Resources/Java/hardware/tools/lm4f/bin/cc3200prog /dev/cu.SLAB_USBtoUART /var/folders/cx/gp5lqpr93z37lg6n7cgjw1m40001kv/T/build5277953067437026045.tmp/Blink.cpp.bin 
Can't open device
Open UART /dev/cu.SLAB_USBtoUART
open UART success
UART Receive Timeout
Read ACK failed
Failed to trigger bootloader
m1959:Energia flety$

rei-vilo, 2015-05-20:
Another issue with uploading: see #624 **CC3200 : "Failed to trigger boot loader" Error When Uploading**

flety-emmanuel, 2015-05-28:
any update on this ? Ideally, someone with access to the loader's code need to buy a cheap siliconlabs USB serial adapter and log what's happening when opening the port. That's what I've been using, even with the launch pad, to formerly test that I could upload some code to the CC3200 without a specific / complicated USB interface (tested on PC only, obviously). I've used one of those
http://www.ebay.com/sch/i.html?_from=R40&_trksid=p2047675.m570.l1313.TR0.TRC0.H0.XCP2102.TRS0&_nkw=CP2102&_sacat=0

thanks

flety-emmanuel, 2015-05-29:
is there a debug / more verbose output of cc3200prog so that I could help diagnosing what is going wrong with the opening of that CP2102 usb serial port ? I really need to get this to work properly

flety-emmanuel, 2015-05-29:
I could even send a USB module to you robertinant, if needed. Let me know.

rei-vilo, 2015-05-29:
Have you tried on the e2e.ti.com forum?

flety-emmanuel, 2015-05-29:
nope. Didn't make sense to me as it's related to a specific upload tool (CC3200prog) which is closed source, which are accessible only (AFAIU) to red bear labs and robertinant. The port itself works great. I can open it with any other software, any serial terminal like cool term, on PC or mac. CC3200prog works on PC too. From the command line (in a terminal, like someone suggested above), it complains that it cannot open the device, but it goes ahead and say UART open success. So access to the HW seems ok (as other software show) but CC3200prog seems to filter the accepted devices, for some reasons. Again, the HW and the serial port itself just work fine. Aside the name filtering in the serial port menu of Energia (already addressed and easy to fix), it's not an energia or even a TI framework problem. it's all in the code of CC3200prog.

rei-vilo, 2015-05-29:
The primary goal of Energia is to support the LaunchPad boards. 

RedBearLab uses a different programmer based on the [K20DX128VLF5](http://www.freescale.com/webapp/search.partparamdetail.framework?PART_NUMBER=MK20DX128VLF5&lang_cd=#General_Information) from Freescale.

As the CC3200 features serial wire debug port (SW-DP) and serial wire JTAG debug port (SWJ-DP) debug access, you or your organisation can use [OpenOCD](http://openocd.org) or the [Uniflash Standalone Flash Tool for TI Microcontrollers (MCU), Sitara Processors & SimpleLink devices](http://www.ti.com/tool/uniflash), instead of  `CC3200prog`, provided your programmer is supported. 

Why can't you for support on the CC3200 (mod) at the http://e2e.ti.com forum?

flety-emmanuel, 2015-05-30:
again, I think you're missing the point. Our goal is to use energia so that the CC3200 board can be programmed on PC and on Mac. OpenOcd aren't included in the toolchain allowing to send the code to the CC3200 by just clicking on the upload button. As far as I know, uniflash isn't available for mac neither. We're trying to have a programming friendly, arduino style for our users, on both platforms. I can currently use my board on PC with CC3200prog *as it is*. Why it won't work on mac is beyong my understanding.
I understand the will to support primarily lauchpads but again, all I'm trying to have is a serial port to upload code. I now feel there are deliberate restrictions. I totally got the RBL board is using a different uploader, basically it's mimicking the FTDI double UART and as they are the authors of the CC3200prog software it's all compatible. So it's sounds like the "official" launchpad and their boards are "accepted" here and others aren't, just because the USB serial port doesn't have the right name or VID.
Again, I'm just looking for an working uploader to be included in the toolchain, so that it's easy for any one to use arduino style programming within a single IDE rather than having to use external tools to upload the code *while it should work as it is* (which is the case on PC).
Again, it's not concerning the CC3200mod chip itself. The issue lies in the CC3200prog. We can program with uniflash too, it's just totally cumbersome and non sense to do that on a mac.

flety-emmanuel, 2015-06-01:
a small update on this thing : on the log above, I figured out what the "can't open device" means there. I've installed the newest version of uniflash and it says something similar but more detailed : it means it didn't find a FTDI USB device, which isn't incompatible with opening the port and trying to upload anyway, which explains why it keeps opening the port (which works). However, it doesn't end well, so maybe it's a question of the timeout on the ACK. 
Still, I don't understand why it works on PC and not on Mac.
The messages and what follows the opening of the port seem different on each OS. Here's the log on my PC.

Opening \\.\COM10
Getting storage list
Bootloader Version: 4
Silicon version ES1.32 or higher
Bootloader version is 2, 1, 4, 0
It's a CC3200 device: PG1.33 or higher
Switch UART pinmux to APPS
Switch to NWP bootloader complete
Load common boot command for PG1.33 or higher
Bootloader version is 2, 0, 4, 0
BlockSize is 4096, number of blocks is 16
Can't open device
erasing 13 blocks starting from  0
erasing file "/sys/mcuimg.bin"
deleting file "/sys/mcuimg.bin"
erase file completed
Downloading file "/sys/mcuimg.bin" with size 26816
.......Download complete

robertinant, 2015-06-01:
"Can't open device" device is that cc3200prog can't find the FTDI device which is used in combination with the jumper from JTAG to SOP2 to put the device into bootloader mode.

The ACK timeout might indeed be to short. I'll send you an updated version to see if a longer ACK timeout works.

rei-vilo, 2015-06-01:
The problem might be Mac OS X and its erratic management of USB instead of `cc3200prog`. Every new release of Yosemite changes the way Mac OS X deals with USB —see the case for mbed. Apple has lost its focus on quality.

So, Monsieur Emmanuel Flety, please provide more information about the exact release of Mac OS X. And please keep in mind Energia is supported by volunteers.

flety-emmanuel, 2015-06-01:
thanks to you both. The OS versions were mentioned in my first post here, we've tried yosemite and 10.9. I'll fetch the exact version # if it helps.
I've got previous experience with USB drivers all along mac os, and I'm indeed very well aware of the current lock trend of signed vs non signed drivers, but like I've mentioned the CP2102 is using signed drivers that are working fine in every software I could test, arduino, coolterm, basically anything serial.

thank you also for for clarifying that the "can't open device" is indeed linked to the automatic set of the flashing mode (which I'm doing manually with a switch, that isn't a problem on our end). So it seems the port is opened on both platform and it might be just a matter of handshakes between the uploader and the board.

I totally got the fact volunteers are dealing with energia. That's actually precisely why I've been selecting this solution, this bonds arduino style of programming, hackers, DIY along with a extreme reactivity of people here or on 43OH to improve the API and classes so that we can use those GREAT chip on our platform without the need to bite the bullet with a low level driver / framework when it's not needed. Just showing my enthusiasm here, not stirring the pot, I just want to push for having energia MORE used because I think it's so great. 
I'm an embedded hw engineer and I've been dealing with low level, asm or C on PIC proc, but with our board, we're promoting empowerment of users who aren't used to low level programming but rather signal processing on computers. The goal is to provide the possibility for them to program on embedded hw with their knowledge in terms of algorithmic without  getting down in the rabbit hole and arduino / energia is so perfect for you.

thanks again for your time !

rei-vilo, 2015-06-01:
You're welcome! 

Yes, please mention the sub-release of Mac OS X Yosemite, as it is critical. 

For the mbed case, 10.10 and 10.10.1 worked, 10.10.2 didn't worked and 10.10.3 worked again.

flety-emmanuel, 2015-06-02:
the mac we're currently trying the CC3200prog version with the longer ACK timeout is yosemite 10.10.3. I need a oscilloscope to check that we do have the port *actually* opened by the uploader app, thought it prints that it's opening the UART with success

flety-emmanuel, 2015-06-06:
it seems that the CP2102 doesn't react to the break signal on mac os only while it works on PC. As it's working with the FTDI and using the same mac API to access the serial port, it must be a driver problem from silicon labs.

robertinant, 2015-06-08:
Thanks for reporting back!

closed 2015-06-08 by robertinant

Parrrk, 2015-11-17:
Anybody success on using CP2102 with CC3200 on MAC OS X?
If it's a problem of SiliconLabs, what can I do for that?
Any suggestion would be appreciated.

rei-vilo, 2015-11-17:
@Parrrk 
Which release of Mac OS X please?

rei-vilo, 2015-11-17:
Similar to #764?

flety-emmanuel, 2015-11-17:
@Parrrk : that will never work on mac OS with the CP2102. The CC3200 expects a break condition on its UART to trigger the booloader mode. The CP2102 driver under mac OS does *NOT* have the break implemented as seen on the silicon labs forums. Use a FTDI if you want it to work on both mac and windows.

Parrrk, 2015-11-17:
Thank you rei-vilo.
I'm in Yosemite ( 10.10.5 ). But it seems like cp2102 has no hope in OS X.
Thank you flety-emmanuel. for your kind explanation of problem.
I read a post about using cp2102 to upload binary in Energia. 
(https://hackaday.io/project/4306-cc3200mod-breakout/log/14436-programming)
But, I think he is using it under Windows.
Thanks anyway..

rei-vilo, 2015-11-17:
Sorry, which board are you using? 

* The LaunchPad CC3200 features a FTDI chip. 
* The RedBearLab CC3200-boards rely on a K20 turned programmer.

I don't know CC3200-based boards with a CP2102. Have you installed and configured the [drivers](https://www.silabs.com/products/mcu/Pages/USBtoUARTBridgeVCPDrivers.aspx).

Mac OS X 10.10 and 10.11 experience issues with USB. 

Please contact [Silicon Labs support](https://www.silabs.com/support/Pages/default.aspx) and/or file a [bug report at Apple](http://bugreport.apple.com).

flety-emmanuel, 2015-11-17:
We have successfully used a CP2102 under windows as the driver implements the break condition that is missing under mac OS. It's a siliconlabs issue, their macos driver is the culprit. I've posted on their forums long ago and never got that issue addressed, therefore moved back to a FTDI uart.