energia/Energia #624: CC3200 : "Failed to trigger boot loader" Error When Uploading

rei-vilo, 2015-05-20:
When uploading to the CC3200 LaunchPad,

```
/Applications/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/cc3200prog /dev/cu.usbserial-cc3200B Builds/embeddedcomputing.bin
```

it fails in 90% of the cases with the following error message:

>
Open UART /dev/cu.usbserial-cc3200B
open UART success
Read ACK failed
Failed to trigger boot loader
>

How to improve the reliability and capability of the upload process?

flety-emmanuel, 2015-05-20:
good point. I think it's due to what robertinant mentioned, a ACK timeout on the UART that is a bit short. The reset of the launchpad seems to be a bit different than with the CC3200 mod. We forced the reset to be a bit faster to match the ACK most of the time but we still have issues about this. I believe that it totally depend on how fast the chip resets, an average has to be found I suppose. I longer timeout is better to match more different cases / board designs.

robertinant, 2015-05-20:
What is the size of the binary?

rei-vilo, 2015-05-20:
The *blinky*
> Estimated Flash:  2976 bytes used

A web-server
> Estimated Flash:  47952 bytes used

flety-emmanuel, 2015-05-20:
similar on my end, tried various example codes of similar size, and up to 101 kb for my "full" program.
Sometimes a fix to this is to keep the CC3200 in reset until energia says uploading, then manually release from reset. Most of the time the auto reset works.
Also, sometimes the upload time is a blitz (even for 100 kb), sometimes it will take ages for the very same binary

robertinant, 2015-09-24:
I have increased the uart receive timeout and this seems to have fixed the issue in most cases. This fix is available in Energia 16. If you are still seeing the issue then please reopen.

closed 2015-09-24 by robertinant

flety-emmanuel, 2015-09-30:
Thanks a lot Robert, that will help for sure !
I'll give a try to energia v16 !

cheers,
Manu

scopedial, 2015-10-13:
I have downloaded energia 16 and I am having the same problem as above. I was uploading just "Blinky".

flety-emmanuel, 2015-10-14:
works for us both on mac and windows

rickenbroc, 2015-10-27:
I also still have the problem too with max osx 10.6.8 both with energia stand alone and CCS cloud

nickstoughton, 2015-11-10:
I still see this issue with Energia 0101E0016 and Mac OS/X El Capitan (10.11). I can upload successfully about 1 time in 100. For the rest, the commonest problem (~90%) is "Failed to trigger bootloader". I also see UART Receive Timeout (~8%) and Timeout during upload (~2%) (?not sure about the spelling of this one). The binary is small (Downloading file "/sys/mcuimg.bin" with size 25776).

Jason0416, 2016-02-04:
CC3200MOD LaunchPad,Stillllllll "Failed to trigger boot loader"

rei-vilo, 2016-02-04:
@Jason0416 Please, which operating system, Energia releases do you use?

robertinant, 2016-02-04:
Have you followed the guide posted here? http://energia.nu/cc3200guide/
Also make sure that you have the drivers installed as per guide here:http://energia.nu/guide/
It would help if you could post the output in the little black window in Energia here.

val123456, 2017-01-19:
Note my daughter is seeing this same error while trying to get it running for a class . . . OS X 10.11.

fletcherbach-temboo, 2017-02-17:
Having the same problem here: 
MacOS 10.12.3 - Retina Macbook Pro - 2015
Energia 1.6.10E18
TI LaunchPadXL CC3200

I followed the launchpad setup guide for MacOS, but might be worth noting that is my first time working with the board and the IDE. 

Console output:
`
Sketch uses 29,848 bytes (11%) of program storage space. Maximum is 262,144 bytes.
Open UART /dev/cu.usbserial-cc3200B
open UART success
Read ACK failed
Failed to trigger bootloader
An error occurred while uploading the sketch
`

rei-vilo, 2017-02-17:
@fletcherbach-temboo 

- Is the LaunchPad configured for upload? 

[image]
 
Most of the times, this message is a false-positive. 

- Does the sketch run on the board?

In that case, you can ignore the message.

fletcherbach-temboo, 2017-02-17:
@rei-vilo Success! 

I jumped TCK to SOP2 per your suggestion and it solved the problem. Thanks for posting the solution and thanks for the quick attention.

rei-vilo, 2017-02-17:
You're welcome!

Full explanation is available at http://energia.nu/cc3200guide/ (mentioned at https://github.com/energia/Energia/issues/624#issuecomment-180018100).