energia/Energia #625: CC3200 : Using Flash to Store Website

rei-vilo, 2015-05-27:
I tested the CC3200 in AP mode. Everything ran fine with easy and fast connection.

However, the Flash still contains TI documents for the CC3200 out-of-the-box demo at 192.1.68.1.1.

I haven't found any mechanism to update the Flash with Energia.

The idea would be to have a feature similar to what Arduino does with the Yún: it uploads the folder `www` into the SD-card. Please refer to section **2. "www" folder** at [Arduino Yún · Intro to web server](http://scuola.arduino.cc/lesson/b4EoRkV/Arduino_Yn_Intro_to_web_server).

A strong limitation: the CC3200 in AP mode is currently limited to one client STA.

spirilis, 2015-05-27:
SLFS lets you read/write the Flash fyi.
But I don't believe you can list contents, in fact there's no in-system way to do so necessarily by security architecture's sake.

However what Energia most certainly does not do atm is configure the NWP's built in webserver.  It's been on my mind for 8 months or so but I got sidetracked recently.

flety-emmanuel, 2015-05-29:
I've used SLFS with great success (thanks again for that lib). While not as practical as drag & drop, you can also make a sketch in energia that will self upload the website by creating the files using SLFS. Or use uniflash by adding a custom file to the upload session