energia/Energia #626: Error on uploading firmatta to tiva C

cjdg, 2015-05-30:
when i load the StandardFirmata i get the next error 

StandardFirmata.cpp.o: In function `setPinModeCallback(unsigned char, int)':
StandardFirmata.cpp:(.text._Z18setPinModeCallbackhi+0x5a): undefined reference to `Servo::attached()'
StandardFirmata.cpp:(.text._Z18setPinModeCallbackhi+0x1ec): undefined reference to `Servo::attached()'
StandardFirmata.cpp.o: In function `sysexCallback(unsigned char, unsigned char, unsigned char*)':
StandardFirmata.cpp:(.text._Z13sysexCallbackhhPh+0x1d6): undefined reference to `Servo::attached()'
collect2: error: ld returned 1 exit status

rei-vilo, 2015-05-30:
Please refer to #384 and #457 and #608.

chintanp, 2015-06-18:
So, there is no solution to the error .. correct ? I read those issues, and couldnt find anything relevant.

rei-vilo, 2015-06-18:
It seems Firmata has been ported to Energia.

chintanp, 2015-06-19:
Does that mean, we dont have to separately download the folder and add it to libraries ? I did as instructed in https://github.com/energia/Energia/tree/master/libraries/Firmata for initialising Firmata example - StandardFirmata, but I got the error as OP.

rei-vilo, 2015-06-22:
If you have a look at the [Boards.h](https://github.com/energia/Energia/blob/master/libraries/Firmata/Boards.h) file, only `__TM4C123GH6PM__` i.e. LaunchPad Tiva C Series is supported, not `__LM4F120H5QR__` i.e. LaunchPad Stellaris. 

Just edit the corresponding line and add the missing board:

```
#elif (defined(__TM4C123GH6PM__) || defined(__LM4F120H5QR__))
```

chintanp, 2015-06-22:
I am using a TM4C123GXL Launchpad, and when I read the boards.h in the firmata library, it has following definitions in TM4C123GXL section

// Tiva C TM4C123G - EK-TM4C123GXL
#elif defined(__TM4C123GH6PM__)
#define TOTAL_ANALOG_PINS       13
#define TOTAL_PINS              41
#define VERSION_BLINK_PIN       40
#define IS_PIN_DIGITAL(p)       ((p)>=2 && (p)<=TOTAL_PINS && (p)!=16 && (p)!=20 && (p)!=21 && (p)!=22)
#define IS_PIN_ANALOG(p)        ((p)>=23 && (p)<=29) || (p==2) || (p==5) || (p==6) || (p==7) || (p==18)
#define IS_PIN_PWM(p)           ((p)==2 || (p)==3 || (p)==4 || (p)==7) || \
								 (p)==14 || (p)==15 || (p)==17 || (p)==19 || \
								 ((p)>=23 && (p)<=26) || (p)==30 || \
								 ((p)>=31 && (p)<=40)
#define IS_PIN_SERVO(p)         (IS_PIN_PWM(p) && (p) >= 0 && (p) < MAX_SERVOS)
#define IS_PIN_I2C(p)           ((p) == 19 || (p) == 38)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p)>=23 && (p)<=26)?(30-(p)):\
								 ((p)>=27&&(p)<=29)?(29-p):\
								 ((p)==18)?(3):\
								 ((p)==2)?(11):\
								 ((p)==5)?(9):\
								 ((p)==6)?(8):(10)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p)

What more can I do ?

chintanp, 2015-06-22:
Actually, I understood what you meant when I saw the changes you had made to boards.h, even using the new version results in the same error, which I present verbatim: 

    StandardFirmata.cpp.o: In function `setPinModeCallback(unsigned char, int)':
    StandardFirmata.cpp:(.text._Z18setPinModeCallbackhi+0x5a): undefined reference to     `Servo::attached()'
    StandardFirmata.cpp:(.text._Z18setPinModeCallbackhi+0x1ec): undefined reference to `Servo::attached()'
    StandardFirmata.cpp.o: In function `sysexCallback(unsigned char, unsigned char, unsigned char*)':
    StandardFirmata.cpp:(.text._Z13sysexCallbackhhPh+0x1d6): undefined reference to `Servo::attached()'
    collect2.exe: error: ld returned 1 exit status

rei-vilo, 2015-06-22:
Are you using the servo library? 

Try and comment the lines related to the servo on the Firmata library.

Also, check `TM4C123GH6PM` or `__TM4C123GH6PM__` is duly defined by Energia with `Serial.println(TM4C123GH6PM);`. 

The Firmata library has been updated for the last 6 months, so I guess it is not widely used and no longer maintained.

chintanp, 2015-06-22:
I am not doing anything with servo. I just want to run the vanilla code as in the example, it just gives me these errors. I tried commenting out the Library: Servo.h, but that creates so many errors, as the code is highly dependent on this library. 

I did Serial.println(__TM4C123GH6PM__); and it worked. 

Sorry to bother u, if you are not the maintainer, but suggest keeping the issue open,

rei-vilo, 2015-06-22:
The Firmata library isn't part of the compiled Energia.

A simple sketch compiled against the Stellaris LaunchPad as well as other boards throws the error

>
sketch_jun22a.ino:1:21: fatal error: Firmata.h: No such file or directory
compilation terminated.
>

```
#include "Firmata.h"

void setup() { }

void loop() { }
```

SO, IMHO, this library is no longer supported.

rei-vilo, 2015-06-22:
Adding the Firmata folder manually to the `Library` sub-folder of the sketchbook makes it available.

Compilation against the Stellaris LaunchPad completes successfully. I can't replicate the issue.

rei-vilo, 2015-06-28:
Renamed as **Support for Firmata Library**

robertinant, 2017-08-30:
This issue was moved to energia/tivac-core#26

closed 2017-08-30 by robertinant