energia/Energia #627: Can we change "private" to "protected" in WiFiClient class for CC3200?

honteng, 2015-06-01:
I'm working on CC3200 SSL connection.

Is it feasible to change "private" to "protected" for WiFiClient class for CC3200?

Here is the gist.
https://gist.github.com/honteng/69460c6b3f25386c940a

I guess this is not harmful to other project.

spirilis, 2015-06-01:
What's the purpose behind that in your example?

honteng, 2015-06-01:
Thank you for your quick response.

We want to release our company service SDK with SSL connection, but we have to pass root CA files to use SSL correctly.

I know still SSL connection implementation in WiFiClient is in progress, but we want to release our SDK before SSL connection implementation is ready in WiFiClient.

If properties are "protected", then our own SDK class can derive from WiFiClient.  And the child class can access to SSL and socket related properties inside the class. Once WiFiClient class implementation for SSL is ready, we will change our SDK.

We don't want to ask developers to apply patch or change WiFiClient code directly only for our service.

honteng, 2015-06-01:
Btw, here is my co-worker's blog post.

http://blog.gaku.net/nitty-gritty-of-ssl-connection-with-cc3200/

Just FYI.

honteng, 2015-06-02:
If you want to see the code, please refer to https://github.com/moderepo/cc3200

honteng, 2015-06-04:
Just curious. Do I need to take an extra action to convey the discussion here? Is there any rule?

spirilis, 2015-06-05:
If you are able to pull down the whole Energia repo, it may be possible to produce a pull request with the change so long as the change is accurate - i.e. applied to cc3200 and the same library for msp430, lm4f, msp432 (for CC3100 support)... I may take a peek at this soon but I just haven't been doing much with energia (or MCUs at all) the past month.

honteng, 2015-06-06:
Actually I tried to pull-req, and push my branch.

But I go the following error.

Energia [change_private_protected●] % git push --set-upstream origin change_private_protected
remote: Permission to energia/Energia.git denied to honteng.
fatal: unable to access 'https://github.com/energia/Energia.git/': The requested URL returned error: 403

It seems like I need permission?

honteng, 2015-06-08:
Sorry, my bad. I will fork first then I will send pull-req.

honteng, 2015-06-08:
https://github.com/energia/Energia/pull/630

honteng, 2015-06-22:
#630 is closed so I close this issue. Thank you for your support.

closed 2015-06-22 by honteng