energia/Energia #645: Missing Libraries in Energia Built with `ant dist`

rei-vilo, 2015-06-24:
When building Energia with `ant dist`, the following libraries...

```
Energia/libraries/CogLCD
Energia/libraries/M2XStreamClient
Energia/libraries/MQTTClient
Energia/libraries/OPT3001
Energia/libraries/PubSubClient
Energia/libraries/Temboo
```

...are missing from the platform folders

```
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/common/ti/runtime/wiring/cc3200/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200emt/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/c2000/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/cc3200/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/lm4f/libraries
/Applications/Energia.app/Contents/Resources/Java/hardware/msp432/libraries
```

If they are common to all platforms, they should be provided under the folder `/Applications/Energia.app/Contents/Resources/Java/libraries`.

I don't know whether the same issue exists for Windows and Linux, hence the `OS Mac OS X` flag.

rei-vilo, 2015-06-28:
Similar issue happens under Windows. See #649.

robertinant, 2015-09-25:
This has been fixed. The emt tree is now on github and a submodule of Energia repo.
To update, in the root of the Energia repository execute:
```
git submodule sync
```
This will sync the submodule URL.

Then to pull the Energia repo + all submodules execute:
```
git pull --recurse-submodules
```
I will update the readme file for the Energia repo.

closed 2015-09-25 by robertinant