energia/Energia #649: Energia MT — Arduino.h: No such file or directory

rei-vilo, 2015-06-26:
Using Energia built with ant dist` after pull-requests #634 and #647, compiling the *Blink* example against the following boards...

* MSP432 EMT
* CC3200 EMT

...fails with error message...

>
Blink.ino:24:21: fatal error: Arduino.h: No such file or directory
compilation terminated.
>

...but succeeds against CC3200 standard = non-EMT.

rei-vilo, 2015-06-26:
The file `Energia.h` is located under `Energia.app/Contents/Resources/Java/hardware/common/ti/runtime/wiring` but not under `Energia.app/Contents/Resources/Java/hardware/cc3200emt` or `Energia.app/Contents/Resources/Java/hardware/msp432`.

rei-vilo, 2015-06-28:
Same issue with file `board.h`.

See #645 and #650.

robertinant, 2015-09-23:
This has been fixed in Energia 16.

closed 2015-09-23 by robertinant