energia/Energia #654: Educational BoosterPack Screen Library // explicit check for board support

adrianF-TI, 2015-06-30:
For each new Energia release, we need to update the explicit list of supported boards. For example, MSP432 is not in the current Screen_HX8353E.cpp file as a supported board, thus compilation fails. However, by simply adding it to the #if defined() section, it works just fine.

adrianF-TI, 2015-06-30:
Also seems to work with MSP430FR6989 LaunchPad on Energia v16

rei-vilo, 2015-07-01:
The `#if defined()` is required to check the LaunchPad has a 40-pin BoosterPack connector, as `_pinDataCommand`uses pin 31.

Now, we could defined another set of variables for each LaunchPad for the BoosterPack capability, for example `HAS_20_PIN_BOOSTERPACK` and `HAS_40_PIN_BOOSTERPACK`. 

The resulting Screen_HX8353E.cpp would be then 

```
#if defined(HAS_40_PIN_BOOSTERPACK) 
// ...
#else
#error Platform not supported
#endif
```

This would ease the test and save updates when new LaunchPads are available.

rei-vilo, 2015-07-01:
See pull-request 7dce178 on #659 Added `HAS_40_PIN_BOOSTERPACK` and `HAS_20_PIN_BOOSTERPACK`.