energia/Energia #655: MSP430FR6989 WiFi examples don't work

adrianF-TI, 2015-06-30:
The Wi-Fi examples seem to compile OK, but they get hung up at WiFi.begin() when using the CC3100 WiFi BoosterPack with the MSP-EXP430FR6989 LaunchPad

StefanSch, 2015-07-01:
Just gave it a try:
If i also connect an USB to the CC3100 BoosterPack and set the jumper on the CC3100 BoosterPack to LDO (J8) then it does work on my desk.

Could this be a problem of the power supply via the launchpad ?

adrianF-TI, 2015-07-01:
Interesting, I'll give it a try....

adrianF-TI, 2015-07-01:
Confirmed – that worked for me as well! I’ll let the MSP430 LaunchPad hardware team know…

Perhaps we need to add some disclaimer on the example sketch comment block to let folks know to try externally powering the BoosterPack if it doesn’t work.

Thanks
Adrian