energia/Energia #668: AXL3xx sketch does not build for msp430g2231

aftisyst, 2015-07-15:
The msp430g2231 ADXL3xx sketch under "07. Sensors" does not successfully link:

**** Build of configuration Debug for project ADXL3xx ****

@ccs/utils/bin/gmake -k all 
Building file: ADXL3xx.cpp
Invoking: GNU Compiler
"@root/energia-0101E0016/hardware/tools/msp430/bin/msp430-gcc" -c -mmcu=msp430g2231 -DF_CPU=1000000L -DENERGIA=16 -DARDUINO=101 -I"@root/energia-0101E0016/hardware/msp430/cores/msp430" -I"@root/energia-0101E0016/hardware/msp430/variants/launchpad" -I"@root/energia-0101E0016/hardware/tools/msp430/msp430/include" -I"@workspace/ADXL3xx" -Os -ffunction-sections -fdata-sections -g -gdwarf-3 -gstrict-dwarf -Wall -MMD -MP -MF"ADXL3xx.d" -MT"ADXL3xx.d" -fmax-errors=10 -o"ADXL3xx.o" "ADXL3xx.cpp"
Finished building: ADXL3xx.cpp

Building target: ADXL3xx.out
Invoking: GNU Linker
"@root/energia-0101E0016/hardware/tools/msp430/bin/msp430-gcc" -mmcu=msp430g2231 -DF_CPU=1000000L -DENERGIA=16 -DARDUINO=101 -Os -ffunction-sections -fdata-sections -g -gdwarf-3 -gstrict-dwarf -Wall -Wl,--gc-sections -L"@workspace/lpmsp430g2231_core/Debug" -Wl,--undefined,main -fmax-errors=10 -o"ADXL3xx.out" "./ADXL3xx.o" -Wl,--start-group -l"lpmsp430g2231_core" -l"m" -Wl,--end-group 
@root/energia-0101E0016/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: ADXL3xx.out section `.rodata' will not fit in region `rom'
@root/energia-0101E0016/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [0000ffe0,0000ffff] overlaps section .rodata loaded at [0000ff8a,00010031]
@root/energia-0101E0016/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom' overflowed by 82 bytes
@root/energia-0101E0016/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/libcrt0.a(_copy_data.o): In function `__do_copy_data':
/home2/kimballr/mspgcc_energia/BUILD/gcc/gcc/../../../gcc/gcc/config/msp430/crt0.S:195: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in ADXL3xx.out
@workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(TimerSerial.o): In function `TimerSerial':
@root/energia-0101E0016/hardware/msp430/cores/msp430/TimerSerial.cpp:90: relocation truncated to fit: R_MSP430_16_BYTE against symbol `vtable for TimerSerial' defined in .rodata._ZTV11TimerSerial section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(TimerSerial.o)
@workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o): In function `pinMode':
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:63: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_sel0' defined in .rodata.port_to_sel0 section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:79: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_dir' defined in .rodata.port_to_dir section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:80: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_ren' defined in .rodata.port_to_ren section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:81: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_output' defined in .rodata.port_to_output section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o): In function `pinMode_int':
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:110: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_dir' defined in .rodata.port_to_dir section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:111: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_ren' defined in .rodata.port_to_ren section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:112: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_output' defined in .rodata.port_to_output section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:128: relocation truncated to fit: R_MSP430_16_BYTE against symbol `port_to_sel0' defined in .rodata.port_to_sel0 section in @workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o)
@workspace/lpmsp430g2231_core/Debug/liblpmsp430g2231_core.a(wiring_digital.o): In function `digitalWrite':
@root/energia-0101E0016/hardware/msp430/cores/msp430/wiring_digital.c:228: additional relocation overflows omitted from the output
collect2: ld returned 1 exit status
gmake: *** [ADXL3xx.out] Error 1
gmake: Target `all' not remade because of errors.

rei-vilo, 2015-07-15:
Not enough memory. Try against the MSP430G2553 instead.

closed 2015-07-15 by rei-vilo

hemangjoshi37a, 2017-05-12:
I am using  MSP430G2553 and getting `additional relocation overflows omitted from the output` .
What does that suppose to mean?? I it means to have less  program memory than what should I do to solve this problem??

rei-vilo, 2017-05-12:
Memory overflow. Use another MCU with a larger memory.

hemangjoshi37a, 2017-05-13:
@rei-vilo , Okay bro. Thank you...