energia/Energia #670: JoystickDrawColor compilation failure for cc3200 with missing header file

aftisyst, 2015-07-15:
The cc3200 JoystickDrawColor sketch fails to compile and reports a missing Screen_HX8464E.h header file.

@ccs/utils/bin/gmake -k all 
Building file: JoystickDrawColor.cpp
Invoking: GNU Compiler
"@root/energia-0101E0016/hardware/tools/lm4f/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m4 -march=armv7e-m -mthumb -fno-exceptions -DF_CPU=80000000L -DENERGIA=16 -DARDUINO=101 -I"@root/energia-0101E0016/hardware/cc3200/cores/cc3200" -I"@root/energia-0101E0016/hardware/cc3200/variants/launchpad" -I"@root/energia-0101E0016/hardware/tools/lm4f/arm-none-eabi/include" -I"@workspace/JoystickDrawColor" -I"@root/energia-0101E0016/hardware/cc3200/libraries/SPI" -Os -ffunction-sections -fdata-sections -g -gdwarf-3 -gstrict-dwarf -Wall -MMD -MP -MF"JoystickDrawColor.d" -MT"JoystickDrawColor.d" -fno-rtti -fmax-errors=10 -o"JoystickDrawColor.o" "JoystickDrawColor.cpp"
@workspace/JoystickDrawColor/JoystickDrawColor.ino:20:28: fatal error: Screen_HX8353E.h: No such file or directory
compilation terminated.
gmake: *** [JoystickDrawColor.o] Error 1
gmake: Target `all' not remade because of errors.

rei-vilo, 2015-07-16:
Be very careful with the analog inputs as the CC3200 only supports up to 1.5 V. 

As the BoosterPack is designed for 3.3 V, you may zombify your CC3200. 

My advice: don't use any BoosterPack with the CC3200.

rei-vilo, 2015-07-16:
The **EduBPMKII_Screen** is not available for the CC3200, for the reason exposed before.

closed 2015-07-16 by rei-vilo