energia/Energia #671: Arrays sketch for TMS320F28027/TMS320F28069 fails compilation with incompatible declaration error

aftisyst, 2015-07-16:
The TMS320F28027 and TMS320F28069 Arrays sketch fails compilation with an incompatible declaration error:

**** Build of configuration Debug for project Arrays ****

@ccs/utils/bin/gmake -k all 
Building file: Arrays.cpp
Invoking: C2000 Compiler
"@ccs/tools/compiler/ti-cgt-c2000_6.4.2/bin/cl2000" -v28 -ml -mt --include_path="@root/energia-0101E0016/hardware/c2000/cores/c2000" --include_path="@root/energia-0101E0016/hardware/c2000/variants/launchpad_f28069" --include_path="@ccs/tools/compiler/ti-cgt-c2000_6.4.2/include" --include_path="@workspace/Arrays" -g --gcc --define=TMS320F28069 --define=F_CPU=90000000L --define=ENERGIA=16 --define=ARDUINO=101 --diag_warning=225 --display_error_number --diag_wrap=off --asm_extension=S --preproc_with_compile --preproc_dependency="Arrays.pp" --diag_error=66 --set_error_limit=10  "Arrays.cpp"
"@workspace/Arrays/Arrays.ino", line 30: error #148: declaration is incompatible with "void timer(unsigned int, void (*)())" (declared at line 213 of "@root/energia-0101E0016/hardware/c2000/cores/c2000/Energia.h")
1 error detected in the compilation of "Arrays.cpp".

>> Compilation failure
gmake: *** [Arrays.obj] Error 1
gmake: Target `all' not remade because of errors.

**** Build Finished ****

rei-vilo, 2015-11-09:
Just replace 

```
int timer = 100;           // The higher the number, the slower the timing.
```

by 

```
int timerms = 100;           // The higher the number, the slower the timing.
```

rei-vilo, 2015-11-09:
See commit eccdf2a.

closed 2015-11-09 by rei-vilo